CongducPham/LowCostLoRaGw #1: Unrecognized transceiver

Limro, 2016-02-25:
I bought a new Raspberry Pi 2, a 8 GB memory card, and the [SX1272 LoRa shield](https://www.cooking-hacks.com/sx1272-lora-shield-for-raspberry-pi-868-mhz).

However I can't get your software to work on it.
I get the following message from a ``sudo ./lora_gateway_pi2`` call:

```
Unrecognized transceiver
...
...
...
... //forever
```

**Installation process**:

* Downloaded and flashed Raspbian on the RPi2
* Did a ``sudo apt-get update && sudo apt-get upgrade -y`` to get all new update for the Pi
* Inserted the board into the Pi
* Did a ``git clone https://github.com/CongducPham/LowCostLoRaGw.git`` in ``~/``
* Because it is the Libelium and does not use inAir9, all lines in ``radio_makefile`` has a ``#`` (commented)
* Build the software with make lora_gateway_pi2:
```
g++ -DRASPBERRY2 -c SX1272.cpp -o SX1272_pi.o
g++ _lrt -lpthread lora_gateway_pi2.o arduPi_pi2.o SX1272_pi2.o -o lora_gateway_pi2
```
* Executed the file with ``sudo ./lora_gateway_pi2`` and received the "unrecognized tranceiver" message.

So I thought something might be wrong with it. Looking into ``SX1272.cpp``, line 145-165, the version of the unit is detected. But ``version`` gets the reading of ``0x00`` - absolutely nothing.

*Perhaps the shield was broken? *
I unmounted the Libelium from the shield, and connected the pins to the RPi. But still the same message.

*The Libelium is broken?*
I connected an oscilloscope to the ``SPI0_SCLK`` (PIN23), but the clock comes in small burst.
*Perhaps it sleeps?*
In ``setLORA()``, line 456-513, the while loop printing the ``...`` message does indeed do a ``delay(x)`` several times. These, and the ``printf("...\n")`` were commented, so the loop could write and read the SPI all the time - but the burst pattern on the oscilloscope did not change.

I tried 3 different RPi2s and two shield with different Libelium - it's all the same.

What can I do about it?
It is like the SPI is broken to the SX1272.

CongducPham, 2016-02-25:
Dear Rasmus,

Could you check the SPI connections? It seems that it may be a wrong 
connection. Or maybe did you enable SPI on the RPI? It should be done 
because it is not done by default.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Limro, 2016-02-25:
So, I read through the [bcm2538.h](https://github.com/CongducPham/LowCostLoRaGw/blob/master/Raspberry/bcm2835.h) file again, and found out I need to download it aswell.

```
wget http://www.airspayce.com/mikem/bcm2835/bcm2835-1.49.tar.gz
tar zxvf bcm2835-1.49.tar.gz
cd bcm2835-1.xx
./configure
make
sudo make check
sudo make install
```

I then enabled SPI - also as default on boot.
Device-tree was already set to loaded, so it was not an issue.

I rebooted the Pi, but the same still occurs.

Do you have other suggestions?

CongducPham, 2016-02-25:
Dear Rasmus,

I don't see why you need to download the bcm2538.h file because it is 
already in the package.

I did have many reliability issues with the Libelium radios. This is one 
reason I'm using instead either the HopeRF or the inAir modules. Is the 
radio module working fine with the standard distribution and examples 
from CookingHack?

best regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Limro, 2016-02-25:
> I don't see why you need to download the bcm2538.h file because it is already in the package.

No, you are correct. It present a small list stating it skipped the packet.

>I did have many reliability issues with the Libelium radios. This is one reason I'm using instead either the HopeRF or the inAir modules. Is the radio module working fine with the standard distribution and examples from CookingHack?

I tried an example from [Cooking-Hacks](https://www.cooking-hacks.com/documentation/tutorials/extreme-range-lora-sx1272-module-shield-arduino-raspberry-pi-intel-galileo/#step4_2) - it worked.
So out of curiosity I tried the LowCostLoRaGw file again **and it now works properly**.

So, it looks like I installed some files/libraries needed for the program to run.

I have a few RPis which needs this installed. I will be trying to see what might be missing from your installation instruction and create a pull-request to the ``readme.md`` if I find a concrete solution.

CongducPham, 2016-02-25:
Great to hear that it is working new.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Limro, 2016-02-26:
I think I located your problem, so I need you to confirm this.

Try to reboot your RPi, and just run the program.
I am rather confident, that the SX1272 is not initialized correct, because it works after I run another program with a different setup. But if I then reboot, your program no longer works.

CongducPham, 2016-02-26:
Dear Rasmus,

I have a gateway based on an RPI2 and I reboot it every time I use it. 
The radio is a Libelium, connected directly. I have no such issue. This 
radio module works fine from the very beginning. Some other modules from 
Libelium have random behavior and I don't know why. I have no issue at 
all with the HopeRF and inAir9. So I don't know exactly what's going on 
with the Libelium modules as I cannot discard completly a hardware issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

simsoooon, 2016-04-22:
Hi, 

I faced a similar problem with the libelium's LoRa modules, and after a lot of chat with them, they told me to remove a resistor on the module and my module work pretty fine now. 
Here is an extract from the mail they sent to me -->" We think the issue is related of a resistor in the LoRa module that causes the malfunction. If you have some soldering skills it's an easy task, you should remove the resistor and shortcircuit the two pads with a tin bridge. We attach you a photo about the resistor."

I hope this can help you or someone else that use Libelium modules and face and 'unstability' issue. 

Regards. 

[image]

CongducPham, 2016-04-22:
> lora_module_modification

Hi Simon, thank you for this information, I'll try it on the 3 modules 
that have never worked in a stable way and will get back to you. If it 
works for me also then I will have 3 more LoRa module. But really, I 
find the modtronix inAir9 module much better and much cheaper.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

simsoooon, 2016-04-25:
Following your advices I've order some Modtronix last week, I'm waiting for them now. 

Regards. 

Simon

Sasgho, 2016-08-17:
hi friend, great project !!!!

I tried to add in addition to the temperature and humidity sensor  the  PIR sensor (external interrupt digital pin 3)... but i can not try to wake up the sleepy mode for arduino ...my idea it's when received the stimulus pir signal the arduino it's activated again for executed any process

i added this code into for control into loop

                               // Allow wake up pin to trigger interrupt on low.
                                  attachInterrupt(1, wakeUp, CHANGE);

but don't work 

for (int i=0; i<nCycle; i++) {  
              
                     #   // Allow wake up pin to trigger interrupt on low.
                                  attachInterrupt(1, wakeUp, CHANGE);
                   
                                 
                                                 
                                #if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_NANO || ARDUINO_AVR_UNO || ARDUINO_AVR_MINI  
                                          // ATmega328P, ATmega168
                                          LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
                                          
                                          //LowPower.idle(SLEEP_8S, ADC_OFF, TIMER2_OFF, TIMER1_OFF, TIMER0_OFF, 
                                          //              SPI_OFF, USART0_OFF, TWI_OFF);
                                
                                    detachInterrupt(1);  
                                #elif defined __MK20DX256__  
                                          // Teensy3.2
                                #ifdef LOW_POWER_HIBERNATE
                                          Snooze.hibernate(sleep_config);
                                #else            
                                          Snooze.deepSleep(sleep_config);
                                #endif  
                                #else
                                          // use the delay function
                                          delay(LOW_POWER_PERIOD*1000);
                                #endif                 
                                          
                                                                                                                              
                                          Serial.print(".");
                                          Serial.flush(); 
                                          delay(10);                        
                                      }
                    
                    delay(50);
              #else
                    
                     // use a random part also to avoid collision
                    Serial.println(lastTransmissionTime);
                    Serial.println("Will send next value in");
                    lastTransmissionTime=millis();
                    delayBeforeTransmit=idlePeriodInMin*60*1000+random(15,60)*1000;
                    Serial.println(delayBeforeTransmit);
                }
              #endif

kenchia10, 2016-10-07:
Hi,
I am currently using a Lora GPS HAT, 433MHz, and a Raspberry Pi3. Once I have followed your instructions, ran sudo ./lora_gateway_pi2 and I got unrecognized transceiver.
Please advise on how to solve the issue. I know I have to edit the lora_gateway.cpp file, but am unclear on what to change. I have enabled SPI using raspiconfig. 
Apologies as my coding is very bad, I just started learning. 

Hope to hear from you soon! Thanks!

CongducPham, 2016-10-07:
Hi, could you check the SPI connection pin? Take a picture if you can.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-10-10:
Hello, 

Thanks so much for your reply! 

I'm not really sure, I just attached the LoRa GPS Hat into the GPIO pins, can be seen in the photo. 

With thanks.  

[image]

CongducPham, 2016-10-10:
> img_2029

Hi, according to the picture on:

http://wiki.dragino.com/images/9/9b/RPi_GPIO.jpg

the LoRa hat is using GPIO6 for CS pin while our library use CE0 (next 
pin).

As indicated on the dradino wiki (see example 4), there are some 
modifications to make it work.

http://wiki.dragino.com/index.php?title=Lora/GPS_HAT

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-10-11:
Hello,
It works! Thank you so much for your help! Especially your prompt replies!

Have a great week ahead

CongducPham, 2016-10-25:
> Pin update for the LoRa transceiver and ATmega328p on the Mini
> wireless anarduino version, ordered after the pin order on the
> transceiver:
> *RFM95W 328p*
> MISO D12
> MOSI D11
>
> SCK D13
> NSS D10
> RESET n/c
> DIO5 D4 Ready
> DIO2 n/c
> DIO1 n/c
> DIO0 D2 (INT0 interrupt pin) Rx done
> DIO4 n/c
> DIO3 n/c

Hi, if you look at the SX1272.h file, you will see at the beginning the 
following lines:

#if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_NANO || defined 
ARDUINO_AVR_MINI || defined __MK20DX256__
#define SX1272_SS 10
#else
#define SX1272_SS 2
#endif

just to test, use only

//#if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_NANO || defined 
ARDUINO_AVR_MINI || defined __MK20DX256__
#define SX1272_SS 10
//#else
//#define SX1272_SS 2
//#endif

because maybe the define variable for the anarduino is not one of the 
mentioned above.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

mehdiBezahaf, 2017-02-14:
Hi Pham,
It is not a question but more a comment that may help someone else. I have a Lora module (RFM95/ SX1276) with a RP3. The module was used in an old project and the GPIO 25 (pin 22) of the RP3 was used for the CS pin, so when I tried to use your code, I have change the code to tell the program that GPIO 25 is used as CS instead of GPIO 8 (pin 24). Even after changing the code, I kept getting Unrecognized transceiver. After reading your comment about the LoRa Hat, I have welded pin 22 and 24 (and put back the code as it was) and the magic happened! everything works perfectly now! 

Any reason why it doesn't work even after changing the code?! 
Thanks anyway.
Mehdi

CongducPham, 2017-02-14:
Hi, I believe it is because pin 25 is not accessed correctly by the 
ArduPi layer whereas pin 24 which is CE0 is referenced as CS and it is 
usually pin 10 on Arduino. So the mapping with ArduPi is correct. This 
is my guess.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ARAVINDAARONE, 2017-04-03:
i tried using sx1272 in communicating with arduino pro mini, with the basic program which is available in the internet i the sites cooking Hacks

and after uploading there occurs an error like Unrecognised transceiver. How can i get rectified from these problems. I checked the wiring connections and all, it seems to be well and good

 and below is the program  

/*  
 *  LoRa 868 / 915MHz SX1272 LoRa module
 *  
 *  Copyright (C) Libelium Comunicaciones Distribuidas S.L. 
 *  http://www.libelium.com 
 *  
 *  This program is free software: you can redistribute it and/or modify 
 *  it under the terms of the GNU General Public License as published by 
 *  the Free Software Foundation, either version 3 of the License, or 
 *  (at your option) any later version. 
 *  
 *  This program is distributed in the hope that it will be useful, 
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of 
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the 
 *  GNU General Public License for more details.
 *  
 *  You should have received a copy of the GNU General Public License 
 *  along with this program.  If not, see http://www.gnu.org/licenses/. 
 *  
 *  Version:           1.1
 *  Design:            David Gascón 
 *  Implementation:    Covadonga Albiñana & Victor Boria
 */
 
// Include the SX1272 and SPI library: 
#include "SX1272.h"
#include <SPI.h>

int e;

char message1 [] = "Packet 1, wanting to see if received packet is the same as sent packet";
char message2 [] = "Packet 2, broadcast test";

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  
  // Print a start message
  Serial.println(F("SX1272 module and Arduino: send packets without ACK"));
  
  // Power ON the module
  e = sx1272.ON();
  Serial.print(F("Setting power ON: state "));
  Serial.println(e, DEC);
  
  // Set transmission mode and print the result
  e = sx1272.setMode(4);
  Serial.print(F("Setting Mode: state "));
  Serial.println(e, DEC);
  
  // Set header
  e = sx1272.setHeaderON();
  Serial.print(F("Setting Header ON: state "));
  Serial.println(e, DEC);
  
  // Select frequency channel
  e = sx1272.setChannel(CH_10_868);
  Serial.print(F("Setting Channel: state "));
  Serial.println(e, DEC);
  
  // Set CRC
  e = sx1272.setCRC_ON();
  Serial.print(F("Setting CRC ON: state "));
  Serial.println(e, DEC);
  
  // Select output power (Max, High or Low)
  e = sx1272.setPower('H');
  Serial.print(F("Setting Power: state "));
  Serial.println(e, DEC);
  
  // Set the node address and print the result
  e = sx1272.setNodeAddress(3);
  Serial.print(F("Setting node address: state "));
  Serial.println(e, DEC);
  
  // Print a success message
  Serial.println(F("SX1272 successfully configured"));
  Serial.println();
}

void loop(void)
{
  // Send message1 and print the result
  e = sx1272.sendPacketTimeout(8, message1);
  Serial.print(F("Packet sent, state "));
  Serial.println(e, DEC);

  delay(4000);  

  // Send message2 broadcast and print the result
  e = sx1272.sendPacketTimeout(0, message2);
  Serial.print(F("Packet sent, state "));
  Serial.println(e, DEC);

  delay(4000);  
}

CongducPham, 2017-04-03:
Please read our README as our modified library is a bit different.

https://github.com/CongducPham/LowCostLoRaGw/blob/master/README.md

I believe your issue is because of the CS pin. It should on pin 10 on 
the pro mini. If you use our lib and our example it should work. 
Otherwise, change the CS pin to 10 in the original libelium lib.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ARAVINDAARONE, 2017-04-04:
Even after setting the SS pin there also occurs the same error like "Unrecognised transceiver"

CongducPham, 2017-04-04:
Please check the SPI connection again, because it is the only reason I 
can see. Also, did you have the 3.3v version of the pro mini?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ARAVINDAARONE, 2017-04-05:
My controller is Arduino Pro mini with 5v and again the same error occurs as "unrecognised transceiver"

CongducPham, 2017-04-05:
Maybe it is the 5v issue, the radio module requires 3.3v not 5v.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ARAVINDAARONE, 2017-04-05:
but in the actual datasheet of sx1272 the supply voltagei is - (-0.5v) to (3.9v)

 and the operating voltage is - (1.8v to 3.7v)  if it be so, then how can i supply 5v to sx1272

CongducPham, 2017-04-05:
You have to supply 3.3v not 5v, this is the issue with your board.

ARAVINDAARONE, 2017-04-06:
sorry sir its my fault, i made the connection in wrong way. Thank you for your help.

lochoang75, 2018-05-06:
![image](https://user-images.githubusercontent.com/30142021/39671820-b200d700-5149-11e8-902a-d918e8907264.png
I get this error when upload code generice sensor to my Uno. I used dragino board with rf98 lora.

CongducPham, 2018-05-14:
Check the CS pin, it should be pin 10 on the Arduino.

AntoineGirafe, 2018-08-06:
I recently bought a new Raspberry Pi 3, a 32 GB memory card, and the SX1272 LoRa shield from libellium with the SX1272 LoRa Module. 

I tried to use your software but it's not working returning :
```
Unrecognized transceiver
...
...
...
... //forever
```

Just after launching the options 1 of the cmd.sh : 
`sudo ./lora_gateway --mode 1
`
I don't know why it's not working I followed "the step by step" tutorial ...
Any clue about this error ?

CongducPham, 2018-08-10:
Currently, the RPI3B+ is nor supported.

regards,

CongducPham, 2018-09-04:
I believe the SX1272 lib is not an issue, it is probably a change in the architecture so lower level lib such as arduPI and bcm2835 may need to be changed.

AntoineGirafe, 2018-09-04:
ok I will take a look ! the changes between RPI2 and RPI3 are : 

- Built in bluetooth and Wi-Fi

- processor 1.2Ghz quad core (300Mhz faster)

- Architecture 64 bits (previously 32 bits) 

- GPU and RAM faster (higher frequency)

Knowing that, do you have any idea of what could cause such problem in these two low level library ? I will investigate and try to improve it. Thank you again for you help !

CongducPham, 2018-09-04:
The RPI3 is fully supported. Our distrib now targets mainly this platform. It is just the RPI3B+, the last one that we still not support or tested.

AntoineGirafe, 2018-09-04:
OHH My bad,  so it's normally working on Raspberry Pi 3 model B .... Great ! 
I will try again and come back to you if I get the same error with the RF module. Thank you !

AntoineGirafe, 2018-09-04:
OK, so I tried again, same error than my first message  

> I tried to use your software but it's not working returning :
> 
> Unrecognized transceiver
> ...
> ...
> ...
> ... //forever
> 
> Just after launching the options 1 of the cmd.sh :
> sudo ./lora_gateway --mode 1

It's like the module wasn't detected... 
I'm using a cooking hacks shield for raspberry and the libellium module SX1272 I really don't know why it's not working with.
Maybe should I try to connect it pin to pin just like in your exemple without a shield ?

CongducPham, 2018-09-04:
You need to know on the cooking hack shield which pin is the CS pin. And yes, it is much better to avoid the shield.

AntoineGirafe, 2018-09-04:
Concerning the CS pin on the shield, it's a direct link from the pin 4 of the SX1272 module pluged on it (DI012/SPI_MISO)  and the pin GPI08 of the Raspberry 3 so it should work...

I will try it without the shield to be sure.

sarra7, 2021-06-19:
am trying to build a single channel gateway using a RPI 3 and an RFM module existing on Lora Radio Node V1.1 using SPI (photo attached)
However, i keep getting the message unrecognized tranceiver- success `
[image]
[image]