CongducPham/LowCostLoRaGw #2: First Setup

raymonvdm, 2016-03-03:
I`m just discovering the existence of LoRa and have bought an RFM95 and connected the following pin-out

`
PI Port  <<----------------->> RFM95 Port

3v3 = VCC
GND = GND

GPIO10 = SPI_MOSI = MOSI
GPIO9 = SPI_MISO = MISO
GPIO11 = SPI_CLK = SCK

GPIO8 = SPI_SEL  = nCS

`

The radio.makefile

`
CFLAGS=-DRADIO_RFM92_95
#CFLAGS=-DRADIO_INAIR9B
#CFLAGS=-DRADIO_20DBM
`

The startup

`
 ./lora_gateway 
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 4
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa Power to x
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
`

How do i find out of the module is actually found / working? I have no working sensor device so it is difficult to test.

Note when i disable the power to the RFM95 i get "Unrecognized transceiver" so something is working. But did i need to set the GPIO pins somewere?

CongducPham, 2016-03-03:
To test you need at least 2 radios, one for the gw (i.e. raspberry) and 
the other for a device that would send to your gw. Difficult to test 
with only 1 radio!

What do you mean by "... set the GPIO pins somewere?". Actually, only 
the SPI pins need to be connected, this is obviously what you did and it 
is working, so no need for other pins.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

raymonvdm, 2016-03-03:
I`m no hardware tech. I just connected the wires as told abov.  I though i needed to setup which GPIO pin has which RFM95 function. So which is PIN is MOSI and which one is MISO and so on. 

I have on RFM95 and a Kerlink Gateway so i should be fine for some testing. But i cannot figure out how to use the PI as a node (yet) So i first try to make it a single channel gateway.

I also had a LoRA Test Sensor but is seems to be as dead as can be. In the beginnen there was a led blinking but now it is completly dead. So this is stopping my first test case.

CongducPham, 2016-03-03:
>
> I`m no hardware tech. I just connected the wires as told abov. I 
> though i needed to setup which GPIO pin has which RFM95 function. So 
> which is PIN is MOSI and which one is MISO and so on.
Yes, SPI pins are already defined by hardware, so no need to assign 
anything.
>
> I have on RFM95 and a Kerlink Gateway so i should be fine for some 
> testing. But i cannot figure out how to use the PI as a node (yet) So 
> i first try to make it a single channel gateway.
The kerlink is running with LoRaWAN. So if you want to test, you can try 
using an Arduino as an end-device and try the lmic-arduino stack to make 
your arduino LoRaWAN with basic functionality. Then, you will not use 
our package anymore.

If you have the kerlink, probably that the best way for you is to use 
the LoRaWAN distrib and use for instance STM32 boards that are supported 
by the Semtech end-node code.
>
> I also had a LoRA Test Sensor but is seems to be as dead as can be. In 
> the beginnen there was a led blinking but now it is completly dead. So 
> this is stopping my first test case.
Sorry I don't use this platform.

Check OpenLora forum or TheThingNetwork for LoRaWAN related issues with 
your kerlink gateway.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-10-13:
Hi, 
As mentioned in other topics, I am using the Lora GPS Hat, 433Mhz. I am unsure on how to edit the code (lora_gateway.cpp)  to work with 433Mhz, understand that the code is mainly for 868 and 900Mhz. 

From what I have read from the code, in the radio.makefile, I have to uncomment the last one. 
#CFLAGS=-DRADIO_RFM92_95
#CFLAGS=-DRADIO_INAIR9B
CFLAGS=-DRADIO_20DBM

But I am unsure about this (lora_gateway.cpp), as my Lora Hat doesnt fall under these:
// please uncomment only 1 choice
//
// it seems that both HopeRF and Modtronix board use the PA_BOOST pin and not the RFO. Therefore, for these
// boards we set the initial power to 'x' and not 'M'. This is the purpose of the define statement 
//
// uncomment if your radio is an HopeRF RFM92W or RFM95W
//#define RADIO_RFM92_95
// uncomment if your radio is a Modtronix inAir9B (the one with +20dBm features), if inAir9, leave commented
#define RADIO_INAIR9B

Am unsure of this too, am using 433Mhz, so do I need to do this: "#define BAND 868"
// please uncomment only 1 choice
#define BAND868
//#define BAND900

Under the lora_gateway.cpp file, I have to "// Select frequency channel". I am using 433Mhz, so do I have to change all the 868 values to 433? 

Not sure about what else to delete in the lora_gateway.cpp file that is redundant. 

Hope to receive your reply for guidance soon! Thank you so much. So sorry for constantly troubling you, I'm very bad at coding, still trying my best to learn.

CongducPham, 2016-10-14:
You can do a quick test. Please read:

https://github.com/CongducPham/LowCostLoRaGw#running-in-433mhz-band

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|