CongducPham/LowCostLoRaGw #4: Data not communicated

sofibiju, 2016-03-16:
I am using Modtronix InAir9 sx1276 chip with ArduinoUNO.
I have connected pins 13 for SCK, 12 for MOSI, 11 for MISO and 2 for CS. Have connected one LoRa module with the temperature sensor and ardino as the end node. Another LoRa module with Arduino as the Gateway.
The code runs fine and the End node shows Packet sent in the Arduino IDE but the data packet is not being received by the Gateway. Please suggest possible solutions.
Thank you

CongducPham, 2016-03-16:
Did you uncomment the line

//#define IS_RCV_GATEWAY

in the gateway code before compiling and uploading to the Arduino acting 
as gateway?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sofibiju, 2016-03-16:
Yes I have uncommented #define IS_RCV_GATEWAY before compiling and uploading.

sofibiju, 2016-03-16:
For the End node, I have uncommented the following:
#define RADIO_RFM92_95
#define BAND900

For the Gateway I have uncommented the following 
#define RADIO_RFM92_95
#define BAND900
#define IS_RCV_GATEWAY

CongducPham, 2016-03-16:
Could you send me the output at the gateway side?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sofibiju, 2016-03-16:
Gateway output
[image]

CongducPham, 2016-03-16:
> capture

Seems that the gateway is in the 868MHz band, listen on channel 10, i.e. 
865.2MHz. Normally to be able to receive from your node it should be on 
channel 5 of 900MHz band, i.e. 913.88MHz.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sofibiju, 2016-03-16:
I am sorry the o/p image I uploaded previously had #define BAND868 uncommented by mistake.
This is the actual image of the o/p with #define BAND900 uncommented
[image]
This is the image of the end node
[image]
Please suggest anything else that I could try.
Thank you.

CongducPham, 2016-03-16:
I've seen that you have very few ram memory left. That may be an issue.

sofibiju, 2016-03-25:
I am using Rapsberrypi2 for gateway now and have connected the InAir9 module to gpio pins.
I have commented all lines in radio.makefile and IS_RECV_GATEWAY and BAND900 in loragateway.cpp.
I have executed the gateway and only get Sx1276 detected as output and nothing else after that. 
 
[image]

joehunt588, 2016-03-25:
can  you show me how you connected pin in raspi?because mine not show any output

CongducPham, 2016-03-25:
You have to run it as root

 > sudo ./lora_gateway_pi2

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2016-03-25:
OK, sorry I didn't notice that you were already in root mode. So I don't 
see immediately why it is not working. Do you still have this issue?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sofibiju, 2016-03-28:
Yes. It is still the same

sofibiju, 2016-03-28:
@joehunt588 joehunt588 
Raspberrypi2 connections
Connect the required SPI pins 
MISO - GPIO09 
MOSI - GPIO10 
SPI_CLK - GPIO11
SPI_nSSEL -GPIO08
VCC - 3.3V and GND to the corresponding pins on the RPI.

sofibiju, 2016-04-01:
I have tried the gateway with an Arduino mega and now the data has been communicated successfully. So memory must have been the issue when run with Arduino Uno. Thanks for your kind support.

closed 2016-04-01 by sofibiju