CongducPham/LowCostLoRaGw #5: lora_gateway_pi2 hangs?

svdgraaf, 2016-03-19:
Hi! First of all, thanks for all the awesome work, I have some issues with the communications though.

I'm trying to setup a GW on a Raspberry Pi 2, and use an Arduino as a node, both use a RFM95 (SX1276).

At first I was stuck with the `make lora_gateway_pi2`, it built, but it doesn't run (empty output, hangs). When I compile the regular version (`make lora_gateway`), the software does run, here is the output:

```
$ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 4
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa Power to x
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
```

I'm not sure if that is what it should do, but when I disconnect the SPI pins, I get an error (`Unrecognized transceiver`), so I think it works :)

On the Arduino side, I also connect the RFM95 over SPI. Same here, if remove the spi connections, it also hangs (`Unrecognized transceiver`), but with it connected, it seems to be working, and I get this output:

```
SX1272 module and Arduino: send packets without ACK
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 21
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 326
(Temp is 159.18
Sending \!#3#21.55567
Real payload size is 13
--> CAD duration 278
OK1
--> waiting for 1 CAD = 16
--> CAD duration 276
OK2
--> RSSI -120
LoRa pkt seq 21
LoRa Sent in 1620
LoRa Sent w/CAD in 1898
Packet sent, state 1
```

Both are set to use the 868 band (Europe), and I've checked the arduino band, it uses `e = sx1272.setChannel(CH_10_868);` as well.

I don't see any output on the gateway side though. Is it absolutely necessary to use the `lora_gateway_pi2`? And if so: any ideas why it hangs?

svdgraaf, 2016-03-19:
I connected the arduino as follows:

```
RST  -> pin 5
VCC  -> pin 9
NSS  -> pin 10
MOSI -> pin 11
MISO -> pin 12
SCK  -> pin 13
```

CongducPham, 2016-03-19:
Given the output, seems that you have an RPI1 and not the RPI2. Then 
there has been as error with the packet sending (state is 1). I think 
that you should remove the RST pin.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

svdgraaf, 2016-03-19:
Hmmm, removing the reset pin didn't really help, I'll try some other stuff on the Arduino, at least I know the rpi part is correct.

btw: you're right about the rpi1, I picked the wrong one from my junkpile ;)

svdgraaf, 2016-03-19:
Got it! I was reading upon the powersaving stuff, and then it hit me. I moved the VCC for the radio from `pin 9` to the `VCC` pin from the Arduino pro mini, and boom! it works! :+1: :smile:

Thanks for the help!

closed 2016-03-19 by svdgraaf

CongducPham, 2016-03-20:
You do have the 3.3V version right? So you take the power from the VCC 
pin when you are powering with USB right?

When powering with USB, it seems that the voltage is not enough for the 
radio to work fine on a digital pin (9 for instance). But, if you inject 
like 5V from any USB plug into the RAW pin, then it is ok. Thus the 
power saving is working when powering from a battery for instance.

good that it works now.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|