CongducPham/LowCostLoRaGw #7: Issue on Nano

andriyadi, 2016-04-11:
I've tried my own code and example code, tried to make sure the cabling is OK, there's an issue on Nano when executing this:
```
sx1272.sendPacketTimeout
```
The device always restarts. Weirdly, the message is managed to arrive on gateway. But I cannot do anything after calling that statement as the board restarts.

While the exact same code runs well on Uno.

Please advice. Thanks.

CongducPham, 2016-04-13:
Be sure that you do not have memory issues. What code are you running on 
the Nano? A simple end-device code?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

andriyadi, 2016-04-13:
Yep, the one that sends temperature. I even hardcode the temperature data, not actually accessing analogRead. Not much changes. 
Andri Yadi

C. Pham.

-- 

|------------ Congduc PHAM - Professor --------------------------|

|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |

|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |

|----------------------------------------------------------------|

|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |

|Avenue de l'Université - BP 1155                                |

|64013 PAU CEDEX, FRANCE                                         |

|                                                                |

|phone:  [33] (0) 5 59 40 75 94                                  |

|fax:    [33] (0) 5 59 40 76 54                                  |

|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |

|----------------------------------------------------------------|