CongducPham/LowCostLoRaGw #8: LoRaActivitySharing.h

andriyadi, 2016-04-19:
If I want to use LAS, I noticed that it will `#include "LoRaActivitySharing.h"`
But seems I can't find that header file. Where can I get it?

Thanks.

CongducPham, 2016-04-19:
Dear Andi,

The LAS service is under development, we are not distributing it yet.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

gmsotavio, 2016-06-18:
Dear C. Pham.
What is LoRa LAS? Could you clarify this to me?

Thans.
O. A. Gomes

CongducPham, 2016-06-18:
Dear Otavio,

LAS stands for LoRa Activity time Sharing. We propose a mechanism for 
device to shared a global pool of radio activity time to ensure quality 
of service. You may know that in Europe, the 868MHz transmission is 
subject to radio duty-cycled. Generally, 1% is allowed, i.e. 36s / hour. 
LAS proposes to exceptionally go beyond by borrowing activity time from 
another device.

We have a prototyping of LAS service but it is not currently distributed 
because it is still in an early stage of developpment.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

norhaneSupcom, 2017-10-18:
hello, 
i want to use LAS, so is the file loraactivitysharing.h distributed or not yet ?