CongducPham/LowCostLoRaGw #9: End device - Multiprotocol Radio Shield not working.

simsoooon, 2016-04-22:
Hello C.Pham, 

I'm following your work since some weeks and I'm currently working on a LoRa solution with a Pi 1B+ gateway and an Arduino Uno rev 3, both with Libelium modules and shields (you may remember me from OpenLora or Cooking Hacks forums).

According to your tutorial here --> http://cpham.perso.univ-pau.fr/LORA/LoRaDevices.html
As I got a Libelium LoRa radio module, an Arduino Uno and  the shield, I told me I will only have to put your code in the Arduino and everything will be ok.
But it don't work out of the box, so I remove the shield and connect the module via SPI pins. And the end-device work. 
As I got the shield, I manage to get it working  by including <arduinoUtils.h> from cooking hacks examples in the SX1272.cpp and editing some of your code to enable the good socket on the shield. 

So I want to know if I do something wrong and miss something or if the sentence " If you have the shield, you don't have to do anything: just connect the radio module in the left socket and you are ready to go to test the code of the provided temperature sensor. " in you tuto is no longer relevant. 

Regards and thank you for your work.

Simon

CongducPham, 2016-04-22:
> As I got the shield, I manage to get it working by including from

Dear Simon,

This is quite strange but many colleagues reported that they did have 
some issue with the multiprotocol shield. Using direct SPI connections 
seems to solve the issue. Howevern this is not systematic and in some 
other case, it is working with the shield.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

simsoooon, 2016-04-22:
Yes, I can confirm that everything is OK with your code when using SPI connections. 

And can you confirm me that there is no support of the Raspberry Shield from Cooking-Hacks --> https://www.cooking-hacks.com/raspberry-pi-to-arduino-shield-connection-bridge
because when I use it, I must call an example from libelium before I can get your gateway working on the RPI, if I don't call any example I get the "Unrecognized Tranceiver
...     ...     ..." Loop

Simon.

CongducPham, 2016-04-22:
>
> Yes, I can confirm that everything is OK with your code when using SPI 
> connections.
Great, you also save the price of the shield!
> And can you confirm me that there is no support of the Raspberry
> Shield from Cooking-Hacks -->
> https://www.cooking-hacks.com/raspberry-pi-to-arduino-shield-connection-bridge
> because when I use it, [...]
I've never use that shield on the RPI so I can't tell. However, a 
colleague also reported that with the shield on the RPI, she had the 
radio module not responding after a couple of hours of operation. 
Without the shield it is working fine.
>
> Simon.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

simsoooon, 2016-04-25:
On my side, no problems with the PI shield, I can get the gateway working all the time. 

I think with the shield you get a better look solution, no wires but as you say there is a price for that.

I will have to determine the one that better suits my needs as I can work with or without. 

Regards

Simon