CongducPham/LowCostLoRaGw #11: Arduino Uno R3 + inAir9: unrecognized transceiver

pdominique, 2016-05-16:
I followed the instructions in the pdf tutorials to setup a GW with an inAir9 with a Raspberry Pi 2. The GW seems to work just fine (at least there's no error when I start it) but the node doesn't. I get an "Unrecognized Transceiver" error.

I'm using an Arduino UNO R3 and an inAir9 module and I used the following diagram to wire everything with the ICSP pins: [image]

|inAir9|UNO|
|---|---|
|VCC|3.3V|
|CLK|SCK (ICSP)|
|MISO|MISO (ICSP)|
|MOSI|MOSI (ICSP)|
|GND|GND|
|CS|Digital Pin 2 (Interrupt 0)|

I'm new to all of this, is my wiring ok?

CongducPham, 2016-05-17:
The pin assignment is correct. Could you check with a voltmeter that you 
do have 3.3V supplied to the radio? Check again the SPI pins, because I 
don't see why it is not working, apart from an error when wiring the pins.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

pdominique, 2016-05-17:
Thanks, it must have been a bad cable because I just rewired everything and it works as expected now. I had to change line 244 in the simple temp example because channel selection is hardcoded for band 868:

```
e = sx1272.setChannel(CH_10_868);
```

closed 2016-05-17 by pdominique

CongducPham, 2016-05-17:
Great!

I will modified simple temp to check for the band, just as in 
Arduino_LoRa_temp code.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|