CongducPham/LowCostLoRaGw #12: Failed to get byte count on serial. RFM95W on RPi3

sagarkar, 2016-05-24:
Hi,

My fist attempt with LoRa, excited to be part of the future.

I got my hand on RFM95W and I have Raspberry Pi 3. Enabled SPI via raspi-config.

I have connected wires as follow

RFM VCC - RPi VCC (pin 17)
RFM GND - RPi GND (pin 25)
RFM MISO - RPi GPIO 9 (pin 21)
RFM MOSi - RPi GPIO 10 (pin 19)
RFM SCK - RPi GPIO 11 (pin 23)
RFM NSS - RPi GPIO 8 (pin 24)

un-commented CFLAGS=-DRADIO_RFM92_95 from radio.makefile
 
`[pi@raspberrypi:~/lora_gateway $ make lora_gateway_pi2
g++ -DRADIO_RFM92_95 -DRASPBERRY -DRASPBERRY2 -DIS_RCV_GATEWAY -c lora_gateway.cpp -o lora_gateway_pi2.o
g++ -c arduPi_pi2.cpp -o arduPi_pi2.o	
g++ -DRASPBERRY2 -c SX1272.cpp -o SX1272_pi2.o
g++ -lrt -lpthread lora_gateway_pi2.o arduPi_pi2.o SX1272_pi2.o -o lora_gateway_pi2
rm -f lora_gateway
ln -s lora_gateway_pi2 ./lora_gateway
`

Then sudo ./lora_gateway_pi2 

`SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 4
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa Power to x
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
Failed to get byte count on serial.
`

 Please guide me, what did I do wrong?

CongducPham, 2016-05-25:
Dear Sagarkar,

Well, it is the first time I see this error. I don't know whether it is 
specific to RPI3 as I didn't test on this new platform. The message 
clearly comes from the OS, and not from the gateway level. We'll have an 
RPI3 and test whether we do have this issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sagarkar, 2016-05-25:
Dear C. Pham,

Thanks for swift response, much appreciated. I since configured the gateway based on [https://github.com/tftelkamp/single_chan_pkt_fwd](https://github.com/tftelkamp/single_chan_pkt_fwd) and it's running fine.

I am still very much interested in your implementation.

Best regards
Swapnil

CongducPham, 2016-05-26:
Hi, could you try again with the new version of lora_gateway.cpp on the 
github? I think that I found a bug that may be the cause of your problem.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sagarkar, 2016-05-26:
Hi, Just a quick feedback, the error is no more. I will test it further please bear with me

sagarkar, 2016-06-07:
Dear Pham,

Looks like you have solved the problem. Gateway works fine on RPi3. Thank you very much your help and support.

closed 2016-06-07 by sagarkar

Sasgho, 2016-08-28:
Hi all
i have problem because when execute make > lora_gateway_pi2 ...nothing happend
only the cursor blinks (have rraspberry pi2 and rfm95w)

any help please !!!!!

CongducPham, 2016-08-28:
You have to type

make lora_gateway_pi2

don't type the >

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2016-08-28:
hi...yes sorry 
typed wrong ..wont to write "make  lora_gateway_pi2"

what it's wrong? because don't show any information ...for example not SX1276 detected ....
I disconnect the rfm95w and no messages detection. 

regards

CongducPham, 2016-08-29:
Well, it seems that you are only compiling the binary, "make 
lora_gateway_pi2" produces the binary, then you have to run the gateway 
binary;:

sudo ./lora_gateway

for instance.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2016-08-29:
when doing sudo ./lora_gateway_pi2 the %activity of cpu is 100% , then run ths process but don't show nothing...i test too with sudo ./lora_gateway and have similar result
instaled the last rasbian 
installed wiringpi

CongducPham, 2016-08-30:
lora_gateway is the gateway program so it is continously listening for 
incoming packet. It is normal that the cpu usage is high. I don't know 
your test setting but if you don't send anything (or don't have the same 
LoRa parameters at the sending side), then the gateway will not receive 
anything.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2016-08-30:
hi again friend

 want to know if is possible comunicate between the arduino one (node transmiter)and  withraspberry pi (gateway, receptor) by rfm95w ...it it possible ?
regards

CongducPham, 2016-08-31:
Yes, we did that.

regards,

MaxWang1994, 2017-04-28:
Hello,

I have running the "single_channel_gateway" in my raspberry pi and it seems to running and receiving some random data. But I can not receive the data I send with the node myself(the node is just next to the gateway). I have researched a lot but still don't know the reason. Could anyone help me? I can't appreciate more.

Regards.