CongducPham/LowCostLoRaGw #13: Arduino only send one message

wakuwaku, 2016-06-09:
Hi,
Thank you for the excellent instructions you have shared. My Raspberry (LoRa Gateway) and Arduino ATmega1280 clone (LoRa end device) has worked in 1 shot combined with NiceRF chip each (from Alieexpress). I don't have a analog temp sensor connected on it, so it should send some random value's which is fine for the test. 

However, when I power on the device with the arduino serial monitor shows I only send 1 message. What I did expect was that it send again after 10 min. But it doesn't. When I reset the arduino, it sends again 1 message and will be quiet forever. On the gateway, I also see the 1 message that is send.

Do you have any idea if this a desired behaviour? 

Sam
Thanks for the good work!.

CongducPham, 2016-06-10:
It is definitely not the normal behavior. You are compiling using 
LOW_POWER right? Try to compile without LOW_POWER and set
delayBeforeTransmit (defined at the end of the example) to 120000 (2mins).

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

wakuwaku, 2016-06-11:
No I did not enabled the low-power feature. I have swapped the atmega1280 clone for a UNO. And now it updates as it should every 10 minutes. I changed the delayBeforeTransmit to 120000 msec (2mins) to do tests. So the problem was the atmega1280 clone 
Thanks you again and keep up the good work!
Sam