CongducPham/LowCostLoRaGw #17: Support for other LoRa Module

rohinrohin, 2016-06-21:
Hello, 

Thanks for this excellent library. This is surely something all LoRa user's will start using. I would like to know if **your library has support for these modules** as I am looking at purchasing the said modules. Any help would be appreciated :)
 - http://www.ebay.in/itm/Semtech-LoRa-SX1276-Data-Radio-Modem-868-Mhz-DRF1278DM-DORJI-/282062587932?hash=item41ac3d881c:g:Pe8AAOSwv9hW5Qfm
 - http://www.ebay.in/itm/Semtech-LoRa-SX1278-Transceiver-Module-433-Mhz-DORJI-/272261926279?hash=item3f64134d87:g:~IEAAOSwJMhXDnYa
 - http://www.ebay.in/itm/Semtech-LoRa-SX1272-Transceiver-Module-866-Mhz-DORJI-/272268279626?hash=item3f64743f4a:g:~IEAAOSwJMhXDnYa

I am still new to this hence I am finding it difficult to see if the above modules are compatible. Note that one of the modules, communicate through UART (Confident that won't be supported). However I would like to know about the other modules.

Sorry for the long post and thanks for your help, many people had these questions as I could see in the comments/forum's so I wanted to ask a comprehensive question. :)

CongducPham, 2016-06-23:
Dear Rohin,

The first one is as you said based on UART so will not work. The other 2 
should be fine, although the 433MHz band has not been fully tested. I 
think the last one will work, as it is also very similar to the NiceRF 
LoRa which has been tested with our lib.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sagarkar, 2016-06-24:
Hey Rohin,

You can go with [this](http://www.ebay.in/itm/RFM98W-433Mhz-20dBm-LoRa-Transceiver-2-piece-/282012478117?hash=item41a940eaa5:g:Z18AAOSwX~dWtDEH).

I can confirm it works perfect with this library.

CongducPham, 2016-06-24:
Thank you for testing. So did you just try by uncommenting the "e = 
sx1272.setChannel(0x6C4000);" line at the end-device side and running 
the gateway with "--freq 433.0" ?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sagarkar, 2016-06-24:
Sorry Pham, I did posted a wrong link. It was suppose to be RFM95W, which I confirm is working

CongducPham, 2016-06-24:
OK so you didn't test with 433MHz band?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sagarkar, 2016-06-24:
No, I did not test with 433 MHz Band.

rohinrohin, 2016-06-24:
@sagarkar and @CongducPham Thanks a lot :)

rohinrohin, 2016-06-25:
Hey @sagarkar, the wait time on the module is more than 15 days. Do you know of any faster ways to obtain such a module in India or any other compatible module?

Thanks.

sagarkar, 2016-06-25:
Hi Rohin,
Aliexpress is other option, however fast delivery is expensive.

closed 2016-06-28 by rohinrohin