CongducPham/LowCostLoRaGw #19: Unable to send and receive messages

rohinrohin, 2016-06-28:
Hey, 

I have setup two LoRa Modules and I am able to configure the module. (state = 0) for all configuration options. However, I am unable to send or receive packets. I am using the code from https://www.cooking-hacks.com/documentation/tutorials/extreme-range-lora-sx1272-module-shield-arduino-raspberry-pi-intel-galileo/#step6_1 (tx and rx code) to only transmit and receive packets in P2P mode between two LoRa's with your library (that was adapted from Libellium). Hence I was assuming that it would work. I am looking to use the LoRa without a gateway and similar to a nRF24 (RF Module). 

Do you know what I am doing wrong? I am using the DRF1272 (SX1272 module from Dorji). 

Any help is appreciated, Thanks.

rohinrohin, 2016-06-28:
Note: I also tried with gateway and simple_temp code from your library. Still no luck! Unable to transmit or receive messages. 
`6538 bytes of free memory.
SX1272 detected, starting
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 4
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa Power to x
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
`
--- gateway

`Simple LoRa temperature sensor
SX1272 detected, starting
...
Get back previous sx1272 config
Using packet sequence number of 15
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 627
(Temp is 306.15
Sending \!#4#21.55567
Real payload size is 13
--> CAD duration 549
OK1
--> waiting for 1 CAD = 62
--> CAD duration 549
OK2
--> RSSI -115
No ACK
Retry
Abort
LoRa pkt seq 16
LoRa Sent in 6964
LoRa Sent w/CAD in 7514
Packet sent, state 1
Reading 443
(Temp is 216.31
Sending \!#4#21.55567
Real payload size is 13
--> CAD duration 549
OK1
--> waiting for 3 CAD = 186
--> CAD duration 549
OK2
--> RSSI -116
No ACK
Retry
Abort
LoRa pkt seq 18
LoRa Sent in 7154
LoRa Sent w/CAD in 7703
Packet sent, state 1
`
--- node

Since it's able to communicate to the board and say that I have configured it properly which means that there is no error in setting up the module.

Also, which #define should I uncomment for my board? Currently for gateway I am uncommenting
`#define RADIO_RFM92_95` and for end device `#define RADIO_20DBM`

CongducPham, 2016-06-28:
I see that you are requesting ack from the gw. try first by not using 
ack (comment #define WITH_ACK). Also simple_temp is using lora mode 1 
while the default gateway uses lora mode 4, so change accordingly either 
the device, or the gw.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

rohinrohin, 2016-06-28:
Thank you so much. I can confirm that this module works with the DORJI DRF1272 board. You can add that as well if you'd like. Also, I recommend you should change the Lora Mode in the examples folder to match with the gateway, as people would like it to be plug and play (atleast for an example). Also, I think you should add a ping-pong example as well so that they can just check basic functionality. Would you like me to do the above and generate a pull request?

rohinrohin, 2016-06-28:
One other thing, do you know how I can operate these module is P2P mode? (without a gateway). I would just like basic tx and rx and all modules that are in the range will listen it. 

Thanks you so much for the prompt replies and this amazing work :)

CongducPham, 2016-06-28:
> Thank you so much. I can confirm that this module works with the DORJI
> DRF1272 board. You can add that as well if you'd like. Also, I
> recommend you should change the Lora Mode in the examples [...]
Yes, you are right, I will change the default mode of the lora gateway 
to mode 1 (the one that can have maximum range) and set in both 
lora_temp and lora_simple_temp the lora mode to be 1.
>
> Also, I think you should add a ping-pong example as well so that they 
> can just check basic functionality.
By ping-pong you mean with an ACK from the GW? Yes you are right a small 
ping-pong example sending "hello world" and receiving the ACK is a good 
idea.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

rohinrohin, 2016-06-28:
Thanks for your help. Any luck on the P2P Question from libellium?  https://www.cooking-hacks.com/documentation/tutorials/extreme-range-lora-sx1272-module-shield-arduino-raspberry-pi-intel-galileo/#step6_1 I would like to run this code (tx and rx) using your library since I can't use libellium's library.

CongducPham, 2016-06-28:
Normally, you should be able to use sendPacketTimeout at the sender side 
and receivePacketTimeout at the receiver in P2P mode.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

rohinrohin, 2016-07-01:
Thanks a ton! There was some issue initially but that helped. Also, can you help me with this problem. The range I am getting is extremely low (a couple metres). The RSSI from a packet that was about 1m away is -114 dBi. That seems too low. 

Setup - 
Mega with source 9V Battery
LCD (16x2)
4x4 Keypad

The LCD and LoRa is connected to the 5V and 3.3V pins respectively.

CongducPham, 2016-07-01:
You are using the Libelium module? Could you send the initialization code for your LoRa module?

rohinrohin, 2016-07-04:
No I am using a DORJI DRF1272 as in http://www.ebay.in/itm/Semtech-LoRa-SX1272-Transceiver-Module-866-Mhz-DORJI-/272268279626?hash=item3f64743f4a:g:~IEAAOSwJMhXDnYa

Initialization Code : 
``` Arduino
// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
//
// uncomment if your radio is an HopeRF RFM92W or RFM95W
//#define RADIO_RFM92_95
// uncomment if your radio is a Modtronix inAirB (the one with +20dBm features), if inAir9, leave commented
//#define RADIO_INAIR9B
// uncomment if you only know that it has 20dBm feature
#define RADIO_20DBM
/////////////////////////////////////////////////////////////////////////////////////////////////////////// 

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
#define BAND868
//#define BAND900
///////////////////////////////////////////////////////////////////////////////////////////////////////////
#define LORAMODE  1
#define node_addr 3

  
#if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_MINI || defined __MK20DX256__ // Nexus board from Ideetron is a Mini
  #define SX1272_POWER 9
  // the Pro Mini works in 3.3V
  #define TEMP_SCALE  3300.0
#else // ARDUINO_AVR_NANO || defined ARDUINO_AVR_UNO || defined ARDUINO_AVR_MEGA2560 
  // the SX1272 will be powered by the 3.3V pin
  #define TEMP_SCALE  5000.0
#endif

void setup() {
 // Power ON the module
  e = sx1272.ON();

  
  // Set transmission mode and print the result
  e = sx1272.setMode(loraMode);
  Serial.print(F("Setting Mode: state "));
  Serial.println(e, DEC);

  // enable carrier sense
//  sx1272._enableCarrierSense=true;

#ifdef BAND868
  // Select frequency channel
  e = sx1272.setChannel(CH_10_868);
#else // assuming #defined BAND900
  // Select frequency channel
  e = sx1272.setChannel(CH_05_900);
#endif

  e = sx1272.setPower('M'); // setting the power at maximum!, still range is extremely low.

  e = sx1272.setNodeAddress(3);
  Serial.print(F("Setting node addr: state "));
  Serial.println(e, DEC);

  
  
  // Print a success message
  Serial.println(F("SX1272 successfully configured"));

}
```

CongducPham, 2016-07-04:
Try using e = sx1272.setPower('x');

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

rohinrohin, 2016-07-04:
When I put it in setPower('x'), the modules do not transmit itself and I am unable to receive data. An interesting thing to note, with antennae and without antennae, the range seems to be the same (3-5m).  I have checked all antennae connections and the middle pin from antennae goes to ant and the GND goes to the other pin. 

I have attached my setup as well. Thanks again for all the help! :+1: 

[image]
[image]
[image]

CongducPham, 2016-07-05:
> img_20160704_132236
> img_20160704_132223
> img_20160704_132231

Well, I know that antenna connection can introduce a lot of attenuation 
if not done properly. Maybe you can just solder a simple wire antenna 
that probably came with the radio module? What kind of antenna did you 
use BTW? Are you sure that it is an 868Mhz antenna and not an 2.4GHz 
antenna from a 802.15.4 module?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

thefreejr, 2016-07-10:
Sx1272 chip has two legs for rf transmition. Pa_boost leg works whith X or x setPower option. Rfo leg works whith L, H, M option. 
Look at your datasheet wich leg connected to antenna. It seems you transmitting on wrong leg whithout antenna.

thefreejr, 2016-07-10:
Or look at your lora module circuit

rohinrohin, 2016-07-14:
Thanks for all your input, since this board was causing me a lot of problems, I have purchased the RFM95W modules. I am however still getting a poor range with this module as well. Maybe some configuration is wrong? I have etched a PCB for almost no attenuation. I am getting a better range (5-6 metre's now). But nothing close to what was advertised, even WiFi or nRF24L01 give's a better range. 

I must be doing something wrong again. :( 

I have attached all screenshots and setup below. and I have uncommented RFM95W in the PING-PONG example. Any help is appreciated. 

My RSSI of a packet which was received so close was around -60. Which is pretty surprising! I should be getting a stronger signal. I am also using an 868 Mhz antenna.

Breakout:
[image]

[image]

Setup: 
[image]
[image]

Sorry about the long post.

arjun1307, 2016-07-28:
I also have the same range issue.Not able to communicate over long distance .could you solve it?if so how .please post your idea

bcsteeve, 2017-01-27:
Did you ever get this resolved?

tusharvb19, 2017-04-25:
Did the range issue resolved???
Whats the practical range of DRF1272F?
Or there is other low cost LoRa better than above?
Thank you.

kunalpawar330, 2019-09-27:
> Hey,
>
> I have setup two LoRa Modules and I am able to configure the module. (state = 0) for all configuration options. However, I am unable to send or receive packets. I am using the code from [...]

Hello,
I have purchased the Dorji sx1276 module I am getting timeout error with the drf tool for configuring the module. I have put the EN pin to gnd still same problem. if you have used it please help 
https://potentiallabs.com/cart/buy-868mhz-915mhz-lora-sx1276-uart-interface-rf-wireless-module-drf1276dm?gclid=Cj0KCQjwn_LrBRD4ARIsAFEQFKvI6z7zKV50jkd8f2JnzxAAl-TLu248TU8WYo4HEVd1VKEpz660qF4aAvOlEALw_wcB