CongducPham/LowCostLoRaGw #20: sx1272 only starts when an hdmi screen is atached to the raspberry

luisastudillo, 2016-07-08:
Hello, i have readed your investigation, and followed your guide about building a lora gateway.
But now i have a problem, maybe silly. because it only works when an hdmi screen is atached to the raspberry pi 2 b.
This is what happen when i connect it only by ssh
[image]

And the loop never ends.
i hope you can advise me about how to solve it
thank you

CongducPham, 2016-07-08:
> image

I see, this is indeed quite strange, I've never seen this before. Could 
you send a picture showing how you connect the radio module? What radio 
are you using?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

luisastudillo, 2016-07-09:
I am using a libelium lora modules(SX1272 chip), but this only happens with
one of them

CongducPham, 2016-07-09:
Well, I had many problems with the Libelium modules and found that their 
behavior can be quite unstable.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

luisastudillo, 2016-07-29:
I give up with Libelium
Could you recommend me a reliable module to use with raspberry?

Thanks for your help

CongducPham, 2016-07-29:
I like the modtronix inAir9: http://modtronix.com/inair9.html

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|