CongducPham/LowCostLoRaGw #21: test module P2P mode with 2 module LoRa Radio Shield for Arduino - 900 / 915 MHz

daomiagi, 2016-07-08:
hello,
I have 2 module  LoRa Radio Shield for Arduino - 900 / 915 MHz. but i haven't gateway. and i want connect them by p2p mode. but I not found a example about that. can you help me that problem?

CongducPham, 2016-07-08:
You can just use one arduino as a gateway by uploading the gateway 
sketch with #define IS_RCV_GATEWAY uncommented. Be sure to use the same 
band on the Arduino-gateway and the other Arduino running for instance 
the Simple_temp sketch.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

daomiagi, 2016-07-08:
ak. i can use one arduino as a gateway. i wil try that.
you have a name as vietnames? you can speak vietnames?

*Đào Văn Quyết*

Email: vanquyet2506@gmail.com

Phone: +84 1648 440 760

*School of Electronic and Telecomunications*

*Hanoi University of Science and Technology*

daomiagi, 2016-07-10:
SX1272 module and Arduino: send packets with ACK and retries
Setting power ON: state 1
Setting Mode: state 1
Setting Header ON: state -1
Setting Channel: state -1
Setting CRC ON: state -1
Setting Power: state -1
Setting node address: state -1
SX1272 initialization failed

sx1272 + arduino + Multiprotocol Radio Shield. but them not act? have problem with my hardware?

CongducPham, 2016-07-10:
Yes, seems that your radio module have some issues.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2016-07-10:
Or, connections are not done properly. Maybe the CS pin. What Arduino 
board are you using?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

daomiagi, 2016-07-10:
i use board arduino uno r3 for. i have a Multiprotocol Radio Shield so i not note pin of sx1272. i use socket 0 ad socket 1 but them no act

daomiagi, 2016-07-11:
i have 2 arduino uno r3 + sx1272 and Multiprotocol Radio Shield and i use
example code: SX_01b_RX_LoRa, SX_01a_TX_LoRa. but in monitor i have:
SX1272 module and Arduino: send packets with ACK
Setting power ON: state 1
Setting Mode: state 1
Setting Header ON: state -1
Setting Channel: state 1
Setting CRC ON: state 1
Setting Power: state 1
Setting node address: state 1
SX1272 successfully configured

Packet sent, state 1
Packet sent, state 1

and no recieve data. my hardware have problem?

*Đào Văn Quyết*

Email: vanquyet2506@gmail.com

Phone: +84 1648 440 760

*School of Electronic and Telecomunications*

*Hanoi University of Science and Technology*

rohinrohin, 2016-07-14:
You should try to manually connect the module without using the socket to the respective pins of the arduino. Try this - 

MOSI - Arduino MOSI
MISO - Arduino MISO
NSS - Arduino Digital Pin 2 
3.3V - Arduino 3.3V
GND - GND

Try any of the examples (say PING-PONG example) and just check if the modules get's configured properly. 

This should be your expected output, if all works well. 
```Setting power ON: state 0
Setting Mode: state 0
Setting Header ON: state 0
Setting Channel: state 0
Setting CRC ON: state 0
Setting Power: state 0
Setting node address: state 0
SX1272 successfully configured```