CongducPham/LowCostLoRaGw #23: support SX1276 or not?

lancezhangsf, 2016-07-21:
I have a sx1276 module, want to use this code ,but  some registers are not different.

only some codes support sx1276?

 #define        REG_RX_PACKET_CNT_VALUE_LSB  	        0x17
#define        REG_MODEM_STAT	  				0x18
#define        REG_PKT_SNR_VALUE	  			        0x19
#define        REG_AFC_FEI	  					0x1A
#define        REG_PKT_RSSI_VALUE	  			        0x1A
#define        REG_AFC_MSB	  					0x1B
#define        REG_RSSI_VALUE_LORA	  			0x1B
#define        REG_AFC_LSB	  					0x1C
#define        REG_HOP_CHANNEL	  				0x1C
#define        REG_FEI_MSB	  					0x1D

datasheet show as follows 
LoRa Mode:                                                                  FSK/OOK Mode:
 0x17     RegPay-loadLength                                        0x17   Reserved17

jarandacu, 2016-10-26:
Hi ! 

According to [http://cpham.perso.univ-pau.fr/LORA/LoRaDevices.html](LINK), the library that is an enhanced version of a library initially developped by Libelium, supports SX1276.

Greetings !

lancezhangsf, 2016-10-27:
thank you very much!!