CongducPham/LowCostLoRaGw #24: Lora Low Range issue in SX1272 module

arjun1307, 2016-07-27:
Hi,
I'm trying to build LoRa communication gateway.For this,I used raspberryPi and Semtech SX1272 LoRa module.I can send and receive the message.But when I tested the range I am getting an extremely low range in Line of Sight.Also, I tested the range by placing two modules in two different room separating a closed door.but I couldn't  get the gateway to communicate.I have read that the Lora can penetrate obstacles.But I couldn't .
The configuration for Lora is as follows:
`void setup()
{
  e = sx1272.ON();
  e = sx1272.setMode(1); 
  e = sx1272.setHeaderON();
  e = sx1272.setChannel(CH_12_868);
  e = sx1272.setCRC_ON();
  e = sx1272.setPower('M');
  e = sx1272.setNodeAddress(3);
  delay(1000);
`}``

I also tried with setPower('x').but  not have any progress .what I am missing .

CongducPham, 2016-07-27:
What is the radio module that you use?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

arjun1307, 2016-07-27:
I am using Semtech SX1272 RF1 [http://au.rs-online.com/web/p/radio-frequency-development-kits/9225873/](url)

CongducPham, 2016-07-27:
I haven't tested this board. I believe its design is a bit special 
because it is intended as a dev board. There is a split TX/RX path that 
should be controlled by an RXTX pin which is not supported by our 
library. Don't know however your issue comes from this or not. This 
board uses the PA_BOOST so setting power should be done with "X". But 
once again, maybe it is the RXTX switch that cause the issue, at both 
sender and receiver.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

arjun1307, 2016-07-27:
Thank you for our reply.

arjun1307, 2016-07-29:
Hi.I could solve it.Now i can send and receive data over long distance with obstacle.Thank you for your support.

sagarkar, 2016-07-29:
Hi Arjun,

Please share it with the community on how you solved it. That is how you give it back to the Open source community :)