CongducPham/LowCostLoRaGw #25: add PIR sensor

Sasgho, 2016-08-17:
hi friend, great project !!!!

I tried to add in addition to the temperature and humidity sensor  the  PIR sensor (external interrupt digital int 3)... but i can not try to wake up the sleepy mode for arduino ...my idea it's when received the stimulus PIR  signal the arduino it's activated again for executed any process

i added this code into for control into loop

                               // Allow wake up pin to trigger interrupt on low.
                                  attachInterrupt(1, wakeUp, CHANGE);

but don't work 

for (int i=0; i<nCycle; i++) {  
              
                     #   // Allow wake up pin to trigger interrupt on low.
                                  attachInterrupt(1, wakeUp, CHANGE);
                   
                                 
                                                 
                                #if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_NANO || ARDUINO_AVR_UNO || ARDUINO_AVR_MINI  
                                          // ATmega328P, ATmega168
                                          LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
                                          
                                          //LowPower.idle(SLEEP_8S, ADC_OFF, TIMER2_OFF, TIMER1_OFF, TIMER0_OFF, 
                                          //              SPI_OFF, USART0_OFF, TWI_OFF);
                                
                                    detachInterrupt(1);  
                                #elif defined __MK20DX256__  
                                          // Teensy3.2
                                #ifdef LOW_POWER_HIBERNATE
                                          Snooze.hibernate(sleep_config);
                                #else            
                                          Snooze.deepSleep(sleep_config);
                                #endif  
                                #else
                                          // use the delay function
                                          delay(LOW_POWER_PERIOD*1000);
                                #endif                 
                                          
                                                                                                                              
                                          Serial.print(".");
                                          Serial.flush(); 
                                          delay(10);                        
                                      }
                    
                    delay(50);
              #else
                    
                     // use a random part also to avoid collision
                    Serial.println(lastTransmissionTime);
                    Serial.println("Will send next value in");
                    lastTransmissionTime=millis();
                    delayBeforeTransmit=idlePeriodInMin*60*1000+random(15,60)*1000;
                    Serial.println(delayBeforeTransmit);
                }
              #endif

best regards

CongducPham, 2016-08-22:
Dear Sasgho,

I didn't try wake up on external interrupt. I believe there are some 
examples on the web using dedicated pin on the arduino for that purposes.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2016-08-22:
thanks so much CongducPham!