CongducPham/LowCostLoRaGw #26: don't work the raspberry pi gateway

Sasgho, 2016-08-22:
hi friends 

when I run # sudo ./lora_gateway_pi2 the activity of cpu is 100% but do not shows nothing .... stays running the process indefinitely.

regards

CongducPham, 2016-08-22:
Well, lora_gateway_pi2 is the gateway, so it is normal that it loops 
indefinitely to wait for incoming packets.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2016-08-22:
thanx a lot...so i don't receive the package then(((

have as node with arduino one with transceiver rfm95w and other extreme raspberry pi with transceiver rfm95w.