CongducPham/LowCostLoRaGw #27: Unrecognised Transceiver in Serial Monitor

abrierley, 2016-08-25:
Hi Congduc,

I followed all instructions to make a IoT Sensor. After checking compile and upload I open serial monitor but instead of the display in your example all I receive is :

429 bytes of free memory.
Unrecognised Transceiver

I am using the InAir9B and I have uncommented #define INARI9B

Could you suggest please anything else I should check?

Regards

Andrew

CongducPham, 2016-08-25:
Hi, double check the SPI connection pins. Especially CS pin. What board are you using?

abrierley, 2016-08-25:
I am using the InAir9B on: Arduino Pro Mini ATmega328 3.3V 8M Compatible Board from Modtronix.

And yes I have checked the pins and i had the Mosi and Gnd reversed, sorry to waste your time with such a rooky error....

Thank you so much for your help

Andrew

closed 2016-08-31 by abrierley