CongducPham/LowCostLoRaGw #28: Code won't compile

abrierley, 2016-08-26:
I managed to compile and install the code yesterday so the code must be okay, today however, I cannot get it compile.

I have tried uninstalling everything and reinstalling the libraries and Arduino and deleted the preferences etc.  All to no avail, this is the message I get from the compiler:

I think the area of concern maybe:  main.cpp:51:1: error: unable to find a register to spill in class 'NO_REGS' 

But I have no idea how to fix it, I have tried multiple versions of Arduino cc 

Linking everything together... "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc"  -Os -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/Arduino_LoRa_temp.ino.elf" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/sketch/Arduino_LoRa_temp.ino.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/libraries/SX1272/SX1272.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/libraries/SPI/SPI.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/libraries/Lowpower/LowPower.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/core/core.a" "-L/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp" -lm
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp: In function 'main':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:51:1: error: unable to find a register to spill in class 'NO_REGS'
 }
 ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:51:1: error: this is the insn:
(insn 945 942 948 72 (set (mem:QI (post_dec:HI (reg/f:HI 32 __SP_L__)) [0  S1 A8])
        (subreg:QI (reg/f:HI 1143) 1)) /Users/andrewdbrierley/Documents/Arduino/Arduino_LoRa_temp/Arduino_LoRa_temp.ino:513 1 {pushqi1}
     (expr_list:REG_ARGS_SIZE (const_int 7 [0x7])
        (nil)))
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:51: confused by earlier errors, bailing out
lto-wrapper: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc returned 1 exit status
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: lto-wrapper failed**
collect2: error: ld returned 1 exit status
Using library SX1272 in folder: /Users/andrewdbrierley/Documents/Arduino/libraries/SX1272 (legacy)
Using library SPI at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
Using library EEPROM at version 2.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM 
Using library Lowpower at version 1.6 in folder: /Users/andrewdbrierley/Documents/Arduino/libraries/Lowpower 
exit status 1
Error compiling for board Arduino Mini.

abrierley, 2016-08-26:
I meant to add, this error applies to the arduino_lora_temp and arduino_lora-simple_temp files but the arduino_lora_ping_pong compiles and loads without issues.

CongducPham, 2016-08-26:
Well, that's weird. Which version of the Arduino IDE are have you been 
trying with then? Indicate also the board that you are using.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-08-29:
I am using the cc version 1.6.11 with a Funduino Pro Mini

abrierley, 2016-08-29:
3.3v 8MHz Version

CongducPham, 2016-08-29:
Hi, so this is weird. You said compilation went fine once and then 
didn't work, with exactely the same code?

In the error message you gave, seems that the linking stage has some 
error, sorry, didn't manage to locate why you have such error so far. 
Will keep searching.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-08-29:
I was able to get it to compile after I commented out as below:

//#ifndef _VARIANT_ARDUINO_DUE_X_
//#define WITH_EEPROM
//#endif
#define WITH_APPKEY
#define FLOAT_TEMP
//#define NEW_DATA_FIELD
//#define LOW_POWER
//#define LOW_POWER_HIBERNATE
#define CUSTOM_CS
//#define LORA_LAS
//#define WITH_AES
#define WITH_ACK

CongducPham, 2016-08-29:
OK, strange because the Funduino Pro Mini is very similar to the Arduino 
Pro Mini so EEPROM should work. You should remove WITH_ACK otherwise the 
end-device will request ACK for every transmission.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-08-30:
I thought I had finally managed to get everything working but although it all appears to be; there is no actual data being received on the pi. I have tried Lora Mode 1 and 4 at both ends, cannot get the software to compile with APPKEY either, even though it did earlier.....

Pi Log after starting with: sudo ./lora_gateway_pi2 --mode 1 |python ./post_processing_gw.py -t -L |python  ./log_gw.py -a 2

2016-08-30T15:52:48.551035> rcv gw output to log (^$): **********Power ON: state 0
2016-08-30T15:52:48.551379> rcv gw output to log (^$): Default sync word: 0x12
2016-08-30T15:52:48.551758> rcv gw output to log (^$): LoRa mode 1
2016-08-30T15:52:48.552059> rcv gw output to log (^$): Setting mode: state 0
2016-08-30T15:52:48.552348> rcv gw output to log (^$): Channel CH_10_868: state 0
2016-08-30T15:52:48.552635> rcv gw output to log (^$): Set LoRa Power to M
2016-08-30T15:52:48.552923> rcv gw output to log (^$): Power: state 0
2016-08-30T15:52:48.553246> rcv gw output to log (^$): Get Preamble Length: state 0
2016-08-30T15:52:48.553558> rcv gw output to log (^$): Preamble Length: 1288
2016-08-30T15:52:48.553855> rcv gw output to log (^$): LoRa addr 1: state 0
2016-08-30T15:52:48.554144> rcv gw output to log (^$): SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge

Nothing happens after this?

Partial log from Serial output of Arduino Pro Mini:

SX1272 module and Arduino: send packets without ACK
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 36
(Temp is 11.60
Sending \!#1#11.60
Real payload size is 10
--> CAD duration 546
OK1
--> waiting for 1 CAD = 62
--> CAD duration 547
OK2
--> RSSI -124
LoRa pkt seq 0
LoRa Sent in 1405
LoRa Sent w/CAD in 1954
Packet sent, state 0
0

Arduino_Lora_temp Sketch:

#define BAND868
//#define BAND900
///////////////////////////////////////////////////////////////////////////////////////////////////////////

//#ifndef _VARIANT_ARDUINO_DUE_X_
//#define WITH_EEPROM
//#endif
//#define WITH_APPKEY  // Sketch will not compile if this is not commented???
#define FLOAT_TEMP
//#define NEW_DATA_FIELD
//#define LOW_POWER
//#define LOW_POWER_HIBERNATE
#define CUSTOM_CS
//#define LORA_LAS
//#define WITH_AES
//#define WITH_ACK

#ifdef WITH_EEPROM
#include <EEPROM.h>
#endif

#ifdef WITH_ACK
#define NB_RETRIES 2
#endif

#define DEFAULT_DEST_ADDR 1
#define LORAMODE 1
#define node_addr 6
#define field_index 3
#define node_addr 10
#define field_index 2
#define node_addr 3
#define field_index 1

closed 2016-08-31 by abrierley