CongducPham/LowCostLoRaGw #30: RPI is not receiving messages from Arduino Pro Mini

abrierley, 2016-08-31:
I am using an RPI3 with InAir9B to receive data from Funduino also with InAir9B
No matter how many combinations of with app key or without  etc the RPI3 just does not seem to receive the data that the Funduino is sending?

The funduino is sending date as per log:

SX1272 module and Arduino: send packets without ACK
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 59
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 39
(Temp is 12.57
Sending \!#3#TC/12.56
Real payload size is 13
--> CAD duration 139
OK1
--> waiting for 1 CAD = 16
--> CAD duration 139
OK2
No ACK
Retry
Abort
LoRa pkt seq 60
LoRa Sent in 10514
LoRa Sent w/CAD in 10653
Packet sent, state 1
Switch to power saving modeSuccessfully switch LoRa module in sleep mode

**************************************************************************************

The RPI3 log:

pi@brierley:~/lora_gateway $ sudo ./lora_gateway_pi2 --mode 4 | python ./post_processing_gw.py -t -L --wappkey | python ./log_gw.py -a 2
will use _2 for post-processing log file
will enable thingspeak support
will log gateway message prefixed by ^$
will check for correct app key NOTE set to [1, 2, 3, 4] 
SX1276 detected, starting.
SX1276 LF/HF calibration
...
rcv gw output to log (^$): **********Power ON: state 0
rcv gw output to log (^$): Default sync word: 0x12
rcv gw output to log (^$): LoRa mode 4
rcv gw output to log (^$): Setting mode: state 0
rcv gw output to log (^$): Channel CH_10_868: state 0
rcv gw output to log (^$): Set LoRa Power to M
rcv gw output to log (^$): Power: state 0
rcv gw output to log (^$): Get Preamble Length: state 0
rcv gw output to log (^$): Preamble Length: 1288
rcv gw output to log (^$): LoRa addr 1: state 0
rcv gw output to log (^$): SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
***************************************************************************************

Arduino Lora temp:

#ifndef _VARIANT_ARDUINO_DUE_X_
#define WITH_EEPROM
#endif
#define WITH_APPKEY
#define FLOAT_TEMP
#define NEW_DATA_FIELD
#define LOW_POWER
#define LOW_POWER_HIBERNATE
#define CUSTOM_CS
//#define LORA_LAS
//#define WITH_AES
#define WITH_ACK

#ifdef WITH_EEPROM
#include <EEPROM.h>
#endif

#ifdef WITH_ACK
#define NB_RETRIES 2
#endif

#define DEFAULT_DEST_ADDR 1
#define LORAMODE  4
#define node_addr 6
#define field_index 3
//#define node_addr 10
//#define field_index 2
//#define node_addr 3
//#define field_index 1

<removed irrelevant code to this issue>

int loraMode = LORAMODE;

#ifdef CUSTOM_CS
unsigned long startDoCad, endDoCad;
bool extendedIFS = true;
bool RSSIonSend = true;
uint8_t SIFS_cad_number;
uint8_t send_cad_number;
uint8_t SIFS_value[11] = {0, 183, 94, 44, 47, 23, 24, 12, 12, 7, 4};
uint8_t CAD_value[11] = {0, 62, 31, 16, 16, 8, 9, 5, 3, 1, 1};
#endif

uint8_t my_appKey[4] = {5, 6, 7, 8};   `also tried {6, 6, 6, 6} and also tried no app_key`
// not used for the moment
uint8_t DevId[4] = {0x01, 0x02, 0x03, 0x04};

abrierley, 2016-08-31:
Issue Resolved by moving Gateway closer to the remote device (Funduino), I will neet to investigate further why the range is so poor, they were only 5 metres apart and each had an antenna fitted. (The antenna is 915MHz but that should not reduce capability by such a large amount)

closed 2016-08-31 by abrierley

CongducPham, 2016-08-31:
I think you have an issue with the radio module connected to the 
raspberry: when

rcv gw output to log (^$): Preamble Length: 1288

is different from 8, it means an error in the radio module.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|