CongducPham/LowCostLoRaGw #31: Output to Thingspeak

abrierley, 2016-08-31:
I cannot get data into thingspeak, only the log number appears on field 7,.

I suspect the reason is because of this formatting ( TC/22.98 )but I am unable to understand in the code where this "TC/" string is coming from ?

--> DATA with_appkey: read app key sequence
app key is  [6, 6, 6, 6]
in app key list
valid app key: accept data
rcv msg to log (\!) on ThingSpeak ( default , 3 ): TC/22.98
ThingSpeak: will issue curl cmd
curl -s -k -X POST --data field3=TC/22.98&field7=48 https://api.thingspeak.com/update?key=7AQYGN....etc

CongducPham, 2016-08-31:
If you have the TC field then it means that you are using the new data 
format and you have to get the latest version of post_processing_gw.py 
(see gw_advanced)

Otherwise, comment //#define NEW_DATA_FIELD

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-08-31:
update to latest post_processing_gw.py 

Now I get this error when I try to run:

will use _2 for post-processing log file
Traceback (most recent call last):
  File "./post_processing_gw.py", line 190, in <module>
    f = open(os.path.expanduser("local_conf.json"),"r")
IOError: [Errno 2] No such file or directory: 'local_conf.json'

Could not try the other option I there was no such command in the file.

Andrew

abrierley, 2016-08-31:
Sorted, thankyou for your help once again.

Copied local_conf.json to lora_gateway directory

closed 2016-08-31 by abrierley