CongducPham/LowCostLoRaGw #32: Band900

abrierley, 2016-08-31:
Hello Congduc,

I wanted to try the Band900 on channel 5 option so changed this in both the lora_gateway.cpp and re made and also in the sketch for the Funduino Pro Mini and reloaded.

After all changes and checking everything matched; the system failed to communicate?

I returned to 868 and it is woking again but I would prefer to use 915 as this is the standard for Australia.

If you could offer some insight please?

CongducPham, 2016-08-31:
Seems that your settings is quite unstable. Can you provide the logs for 
the 915MHz band test?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-08-31:
I don't have the log for the serial output from the Arduino but I can repeat the test.

Here is the RPI log for that part of the test:

2016-08-31T13:13:09.318675> rcv gw output to log (^$): **********Power ON: state 0
2016-08-31T13:13:09.318905> rcv gw output to log (^$): Default sync word: 0x12
2016-08-31T13:13:09.319132> rcv gw output to log (^$): LoRa mode 4
2016-08-31T13:13:09.319358> rcv gw output to log (^$): Setting mode: state 0
2016-08-31T13:13:09.319590> rcv gw output to log (^$): Channel CH_5_900: state 0
2016-08-31T13:13:09.319857> rcv gw output to log (^$): Set LoRa Power to M
2016-08-31T13:13:09.320091> rcv gw output to log (^$): Power: state 0
2016-08-31T13:13:09.320363> rcv gw output to log (^$): Get Preamble Length: state 0
2016-08-31T13:13:09.320621> rcv gw output to log (^$): Preamble Length: 8
2016-08-31T13:13:09.320880> rcv gw output to log (^$): LoRa addr 1: state 0
2016-08-31T13:13:09.321141> rcv gw output to log (^$): SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
2016-08-31T15:12:46.231395> will enable thingspeak support
2016-08-31T15:12:46.233258> will log gateway message prefixed by ^$
2016-08-31T15:12:46.233421> will check for correct app key
2016-08-31T15:18:53.443616> will enable thingspeak support
2016-08-31T15:18:53.449415> will log gateway message prefixed by ^$
2016-08-31T15:18:53.449583> will check for correct app key
2016-08-31T15:18:55.905406> SX1276 detected, starting.
2016-08-31T15:18:55.905818> SX1276 LF/HF calibration

CongducPham, 2016-08-31:
Looks good.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-09-02:
I suspect a dry joint on the GRD pin was giving me unnecessary errors Grrrr.

Re-soldered today and a lot of my issues have gone, now working at 915 sending temp through to Thingspeak and everything in the world is beautiful.  Here is my thingspeak channel:

[https://thingspeak.com/channels/148372]

closed 2016-09-02 by abrierley