CongducPham/LowCostLoRaGw #33: Strange problem running without ACK

abrierley, 2016-09-02:
If I run my Arduino_LoRa_temp with #Define WITH_ACK the Sketch runs properly but If I try to comment it out as you suggested in an earlier issue I get the following errors when I try to compile:

"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc"  -Os -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/Arduino_LoRa_temp.ino.elf" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/sketch/Arduino_LoRa_temp.ino.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/libraries/SX1272/SX1272.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/libraries/SPI/SPI.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/libraries/Lowpower/LowPower.cpp.o" "/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp/core/core.a" "-L/var/folders/qd/18nncqg504bbytpfv1dvlmvw0000gn/T/build40b120d426dd3e9d1568e656846a01e9.tmp" -lm
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp: In function 'main':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:51:1: error: unable to find a register to spill in class 'NO_REGS'
 }
 ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:51:1: error: this is the insn:
(insn 988 985 991 76 (set (mem:QI (post_dec:HI (reg/f:HI 32 __SP_L__)) [0  S1 A8])
        (subreg:QI (reg/f:HI 1165) 1)) /Users/andrewdbrierley/Documents/Arduino/Arduino_LoRa_temp/Arduino_LoRa_temp.ino:563 1 {pushqi1}
     (expr_list:REG_ARGS_SIZE (const_int 7 [0x7])
        (nil)))
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:51: confused by earlier errors, bailing out
lto-wrapper: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc returned 1 exit status
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status

CongducPham, 2016-09-02:
Hi, that's really strange. I cannot reproduce your error, I have no 
compilation errors in both cases. Will try to investigate further.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-09-02:
If that is the case then logic dictates that it is something I have done, just seems strange that it would compile with it included but not when it is commented out...  I shall go bracket searching.....

closed 2016-09-02 by abrierley