CongducPham/LowCostLoRaGw #34: Send Data from the Gateway to the Node

keepexploring, 2016-09-11:
Hi,

I've followed through the tutorials and managed to get data flowing from a node to a gateway. Is it possible to send data from the gateway to a node? Perhaps just after the node sends data to the gateway? Does anyone know would you go about doing this? Is there an existing set of functions to do this? I imagine you would need a temporary database on the gateway side which holds the messages until a node sends data and then gets this data and sends to the node, this could be check through the post_processing script, but then how do you send data from the gateway to a node? Is there an existing function that you could call through the post processing script to do this?

Thanks so much!

Joel

abrierley, 2016-09-12:
I am no expert Joel, but I believe in this scenario the nodes are are working in Class A so there is only a short window after transmit from the node to receive a signal, which I guess is how the ACK signal is coming back, maybe check on the gateway where the ACK signal is being sent from and put your message in there?

keepexploring, 2016-09-15:
Thanks so much for this. After receiving a message I'm going to turn the mode into receive mode for 10 second and listen. I'll then send a message from the gateway. Pham very kindly emailed me some code to do this, which I understand he is going to be putting into the github repository soon.

Thanks so much for your help.

kenchia10, 2016-10-11:
Hi,
Is it possible to send data from one node to the another via the gateway? So the whole process would be Node A>Gateway>Node B. 
Would appreciate the help. Thank so much again!

CongducPham, 2016-10-11:
It is not implemented, but we have a new downlink feature that has been 
added, so look at the github. The downlink feature can allow the 
processing stage to initiate a downlink transmission.

The downlink framework can be used for you to implement the behavior 
that you describe (post_processing can initiate a downlink transmission 
to B if node A send a message with the indication that it has to be 
forwarded to B). However, in case of low-power mode, where node B is 
sleeping, it has to know when to wake up to listen, which is costly.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|