CongducPham/LowCostLoRaGw #35: How do I send second field?

abrierley, 2016-09-12:
Hi Congduc,

I have added the following to my Arduino Project to measure the voltage on Pin A2 (Via a voltage divider) in order to measure the battery in my Caravan so it will warn me if the voltage falls below 11v:

int sensorValue = analogRead(A2);
float voltage = sensorValue * (3.3 / 1023.0);
Serial.print("Voltage is ");
Serial.println(voltage);

How do I get the value into the send string, sorry to ask but I am confused as to how this section of the code works?

#ifdef NEW_DATA_FIELD
  r_size = sprintf((char*)message + app_key_offset, "\\!#%d#TC/%s", field_index, float_str);
#else
  // this is for testing, uncomment if you just want to test, without a real temp sensor plugged
  //strcpy(float_str, "21.55567");
  r_size = sprintf((char*)message + app_key_offset, "\\!#%d#%s", field_index, float_str);
#endif

#else

#ifdef NEW_DATA_FIELD
  r_size = sprintf((char*)message + app_key_offset, "\\!#%d#TC/%d", field_index, (int)temp);
#else
  r_size = sprintf((char*)message + app_key_offset, "\\!#%d#%d", field_index, (int)temp);
#endif
#endif
#endif

  Serial.print(F("Sending "));
  Serial.println((char*)(message + app_key_offset));

  Serial.print(F("Real payload size is "));
  Serial.println(r_size);

  int pl = r_size + app_key_offset;

Regards

Andrew

CongducPham, 2016-09-12:
> Hi Congduc,
>
> I have added the following to my Arduino Project to measure the
> voltage on Pin A2 (Via a voltage divider) in order to measure the
> battery in my Caravan so it will warn me if the [...]
Just before these lines, you have in the example:

ftoa(float_str,temp,2);

so for you, just do:

ftoa(float_str,voltage,2);

the node will send (assuming voltage=3.3 for instance)

\!#4#3.3

and that will be processed by the post_processing_gw.py
>
> How do I get the value into the send string, sorry to ask but I am 
> confused as to how this section of the code works?
>
> #ifdef NEW_DATA_FIELD
> r_size = sprintf((char/)message + app_key_offset, "\!#%d#TC/%s", 
> field_index, float_str);
> #else
> // this is for testing, uncomment if you just want to test, without a 
> real temp sensor plugged
> //strcpy(float_str, "21.55567");
> r_size = sprintf((char/)message + app_key_offset, "\!#%d#%s", 
> field_index, float_str);
> #endif
>
> #else
>
> #ifdef NEW_DATA_FIELD
> r_size = sprintf((char/)message + app_key_offset, "\!#%d#TC/%d", 
> field_index, (int)temp);
> #else
> r_size = sprintf((char/)message + app_key_offset, "\!#%d#%d", 
> field_index, (int)temp);
> #endif
> #endif
> #endif
>
> Serial.print(F("Sending "));
> Serial.println((char*)(message + app_key_offset));
>
> Serial.print(F("Real payload size is "));
> Serial.println(r_size);
>
> int pl = r_size + app_key_offset;
>
> Regards
>
> Andrew
>
> <https://github.com/CongducPham/LowCostLoRaGw/issues/35>, or mute the 
> thread 

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-09-12:
That did work but now it just sends the voltage, I wanted to send the temperature and the voltage if possible?

CongducPham, 2016-09-13:
In this case you have to use the NEW_DATA_FIELD format and do something 
like:

ftoa(temp_str,temp,2);

ftoa(voltage_str,voltage,2);

r_size=sprintf((char*)message+app_key_offset,"\\!TC/%s/VCC/%s",temp_str,voltage_str);

and use the post_processing_gw.py script in the gw_advanced folder that 
has been improved to support this data format. You can actually add more 
nomenclature code (e.g. TC, VCC, HUM, CO,...) as you want depending on 
the meaning you want to give to the sensed value.

If you want to do that properly, you can also write some kind of generic 
sensor C++ class which stores both a nomenclature and the sensed value, 
and then loop over all the connected sensor. Something like:

       char final_str[50] = "\\!";
       char aux[6] = "";

       for (int i=0; i<number_of_sensors; i++) {

           if (sensor_ptrs[i]->get_is_connected() ) {

               ftoa(aux, sensor_ptrs[i]->get_value(), 2);

               if (i==0) {
                   sprintf(final_str, "%s%s/%s", final_str, 
nomenclatures_array[i], aux);
               }
               else {
                   sprintf(final_str, "%s/%s/%s", final_str, 
nomenclatures_array[i], aux);
               }
           }
       }

       r_size=sprintf((char*)message+app_key_offset, final_str);

Then, if you use ThingSpeak, look in post_processing-gw.py for the 
following line:

thingspeak_uploadSingleData(data, second_data)

comment it, and uncomment the next line:

#thingspeak_uploadMultipleData(data_array)

This will upload all the value on the default ThingSpeak channel, 
starting by field index 1. So here temperature will be displayed on 
field 1 and voltage on field 2. You may consider using GroveStream 
platform instead, and in this case, the nomenclature code that you give 
will be preserved, which is much more convenient.

hope that help,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

abrierley, 2016-09-13:
Thank you so much once again for your help. I will try and update

abrierley, 2016-09-13:
This is all working great now, thank you. I will investigate the generic class tomorrow but for now all is good. 

For your information Congduc, I am using your solution to monitor the temperature and Voltage of my Deep Cycle batteries in my Caravan as I am worried about them overheating or over charging etc as a result of the Solar Cell that are providing a constant charge.

Next I would like to add a motion sensor to detect theft and I have a few other ideas also, but all you hard work has made it possible, not to mention everything I have learned in the last 3 weeks.

Last but not least, it took me an age today to find information on what the %d and %s etc meant.

best regards,

Andrew

closed 2016-09-13 by abrierley