CongducPham/LowCostLoRaGw #41: Power Settings causing problems

risingmole, 2016-10-01:
Hi there, 

I am using a basic setup with a RPi3 as the gateway and an arduino acting as a node. I am using RFM95W modules. 

I have managed to get both the **simple_temp** and **ping_pong** test codes running, however they only seem to work when I force sx1272.setPower('L') on the arduino to 'L'. If leave it as 'x' or "H' or 'M' then no messages are able to be transmitted/received.

Any advice as to why this might be occurring? Particularly as in the long run I would like to use the higher power modes..

Cheers, 

R

CongducPham, 2016-10-01:
Hi, strange indeed because the RFM95W uses the PABOOST pin so "x" is 
needed. "L" is very low power so I can't really explain this behavior. 
Try checking the received SNR and RSSI. Also, if using mode lora mode 1 
(BW=125, SF=12) keep some distance (2m) between the sender and the gw.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

risingmole, 2016-10-03:
Ok thanks, I have now moved the modules further apart. One is currently upstairs, one downstairs in a timber house (approx 5m apart). Now when the arduino (node) is set to **L**, no messages are received by the gateway (RPi).  If I set the power (on the node) to **'x'** then the gateway is able to receive the ACK request, but the node is not receiving any pong back. 

This is the output from the RPi : 

> `--> CAD duration 201
OK1
--> RSSI -125
^$ACK requested by 8
--- rxlora. dst=1 type=0x18 src=8 seq=18 len=4 SNR=7 RSSIpkt=-80 BW=125 CR=4/5 SF=12
^p1,24,8,18,4,7,-80
^r125,5,12
^t2016-10-03T01:39:47.710
▒▒Ping
`

What do you make of the RSSI/SNR values? I am not really sure what to expect?

CongducPham, 2016-10-03:
The uplink SNR is good (7). Do you have the same radio module for your 
RPI gateway? Did you check for the RPI that the radio.makefile file have 
the correct line uncommented for your RFM95W? Like this:

CFLAGS=-DRADIO_RFM92_95
#CFLAGS=-DRADIO_INAIR9B
#CFLAGS=-DRADIO_20DBM

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

risingmole, 2016-10-03:
Hi again, 
Thanks for the help! I can confirm that I have the same modules for both gateway and node. Also I have the radio.makefile correctly uncommented as you have shown. 

The thing that is really bugging me is that I managed to get a successful connection at one point today using the 'x' and 'X' power modes. Once I switched off the modules to try moving even further away (I had full success in my backyard and needed to move further) i have since not achieved a pong reception by the node. 

At present the node's output loops through 

> Sending Ping
> Packet sent, state 1

However, the request is definitely still getting through to the gateway: 

> --> CAD duration 201
OK1
--> RSSI -125
^$ACK requested by 8
--- rxlora. dst=1 type=0x18 src=8 seq=13 len=4 SNR=6 RSSIpkt=-74 BW=125 CR=4/5 SF=12
^p1,24,8,13,4,6,-74
^r125,5,12
^t2016-10-03T11:48:19.928
▒▒Ping

Any suggestions as to whether it is more likely an issue arising from the gateway - i.e. not transmitting with sufficient power? Or with the node? i.e. not listening at the right time? I have tried tracing through the code to identify what could be the cause of the **state 1 ** error message, but unsuccessful at present.

CongducPham, 2016-10-03:
sometime, at the receiver side, the curent limitation may make the radio 
module to fail. how do you power your radio module? I guess with a pin, 
I don't remember what board you are using as devices.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

risingmole, 2016-10-04:
Ok, thanks, I will try turning down the Receiver gain. On the receiver end (for the node which is sending pings and not receiving pongs) it is an arduino - the rfm95w is being powered by the 3.3V output.