CongducPham/LowCostLoRaGw #42: CH_12_900

risingmole, 2016-10-03:
Hi there, 

It appears that the sx1272.cpp file in the arduino folder is missing a case for the 915MHz band in the isChannel() function. This causes an issue whereby the setChannel state returns an error - since channel 12 is not recognised by isChannel().

The follow needs to be added into isChannel (I believe)
`case CH_12_900:`

Cheers

risingmole, 2016-10-03:
It is also missing from the RPi sx1272.cpp file.

CongducPham, 2016-10-03:
Thanks for pointing out this issue, I will correct the bug asap.

But, as indicated on the github, the required frequency hopping 
mechanism is not implemented.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

risingmole, 2016-10-03:
Great! Thanks very much!