CongducPham/LowCostLoRaGw #43: Unrecognized transceiver

juvanon, 2016-10-17:
Hello, 
I have bought a libellium sx1272, and try to make it work with an arduino DUE. But no way. It keeps on writing on the serial interface. Unrecognized transceiver and for ever ...
Spi connection are ok, I plugged the resetpin. Nothing change.
Do you have a good advise but buying anything else ?

Thanks

CongducPham, 2016-10-24:
On the due, the spi pins must be taken from the ICSP header. Please check these connections. Is the CS pin connected to digital pin 2 of the Arduino Due?

Also, I had a lot of issues with the Libelium SX1272 and I do not recommend these radio chips. The inAir modtronics or the HopeRF RFM95W are much more reliable.

regards,

CongducPham, 2016-10-25:
On the due, the spi pins must be taken from the ICSP header. Please 
check these connections. Is the CS pin connected to digital pin 2 of the 
Arduino Due?

Also, I had a lot of issues with the Libelium SX1272 and I do not 
recommend these radio chips. The inAir modtronics or the HopeRF RFM95W 
are much more reliable.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|http://liuppa.univ-pau.fr/        |http://www.univ-pau.fr    |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr        http://www.univ-pau.fr/~cpham     |
|----------------------------------------------------------------|

liuxiaodong528, 2017-05-19:
@CongducPham I use SX1272 when the module is connected to VCC3.3 and GND is displayed as “Simple LoRa ping-pong with the gateway
Arduino UNO detected
Unrecognized transceiver”

However, when the GND is broken can be identified and used, which is why?
Thanks