CongducPham/LowCostLoRaGw #44: Compile error

vrvoicu, 2016-10-19:
I'm new to this so excuse my ignorance. I've followed your tutorial and I have this problem. How can I solve it?

/home/victor/.arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino/main.cpp: In function 'main':
/home/victor/.arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino/main.cpp:51:1: internal compiler error: Segmentation fault
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: /usr/local/arduinoIde/hardware/tools/avr/bin/avr-gcc returned 1 exit status
/usr/local/arduinoIde/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status

x893, 2016-10-19:
Please submit a full bug report,
with preprocessed source if appropriate.
See http://gcc.gnu.org/bugs.html for instructions.

CongducPham, 2016-10-24:
I think that the newest version of Arduino IDE have some issues. In the board manager, install or select 1.6.9 for the AVR boards libraries.

regards,

CongducPham, 2016-10-25:
I think that the newest version of Arduino IDE have some issues. In the 
board manager, install or select 1.6.9 for the AVR boards libraries.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|http://liuppa.univ-pau.fr/        |http://www.univ-pau.fr    |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr        http://www.univ-pau.fr/~cpham     |
|----------------------------------------------------------------|

jarandacu, 2016-10-26:
Hi ! 

I had the same problem, but i fixit installing the 1.0.6 version of arduino and using this file: Arudino_LoRa_Simple_temp.

Bye !