CongducPham/LowCostLoRaGw #45: Send Data to Gateway from Windows Computer without Arduino

kenchia10, 2016-10-24:
Hello,

Was wondering if you have the code to send data to the gateway from a Windows Computer. Understand that your project is using arduino, but I currently do not have it and hopefully I do not have to use it.

Thanks so much!

kenchia10, 2016-10-24:
In other words, connecting the node(windows computer) to the gateway. I have a 433Mhz Transceiver with a serial converter attached to the computer. 

Would appreciate your guidance. Thanks!

CongducPham, 2016-10-24:
So you can probably control the radio module with serial command using a serial tool.

regards,

CongducPham, 2016-10-25:
So you can probably control the radio module with serial command using a 
serial tool.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-10-26:
Hi, 
I managed to get an arduino, I tried the Ping-Pong program, uploaded it into the arduino, i opened the serial monitor and it showed unrecognized transceiver. 
Am using a 433Mhz Transceiver, SX1278. I connected VCC, GND, MOSI, MISO, CLK to the arduino like your tutorial. 

Would appreciate your advice greatly, thanks!

kenchia10, 2016-10-26:
Thanks so much btw for all your help so far! =)

CongducPham, 2016-10-27:
Hi, what is the pin you use on the Arduino for the CS? What arduino 
board are you using?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-10-28:
Hello,

I used Pin 10, I think in your tutorial you said 2 or 3, I can't remember, but I've tried that too and it doesn't work.

CongducPham, 2016-10-28:
What is your arduino board? For small size form factor, you should use pin 10. For Uno size, use pin 2.

kenchia10, 2016-11-15:
Hello,

Sorry for the delayed reply. Am using Uno. Would like to get the uno working first before changing it to the Pro Mini 5V. 

Appreciate your help. Thanks!

kenchia10, 2016-11-15:
I initially thought there was a problem with my transceiver, thus I took some time to test the transceiver. Managed to get the transceiver to work with a code using the SX1278.h library. So I am back here to try to no avail. I still got unrecognized transceiver with my SX1278 transceiver. It's a china clone that works fine. 

With the uno, I have tried changing the CS to pin 2 as well, initially pin 10. But to no avail. 

Have tried using the Pro Mini 5V, but it does not work too. I got State 3, State 2, when the packet was sent, understand that that means that the message was not sent properly. 

The gateway is up and running using the LoRa GPS Hat 433Mhz with your help previously, thanks again for that. So the gateway shouldn't be the problem. 

Why I chose to debug with Uno as it can Serial.Print the stuff in the void setup, with the Pro Mini, not sure why it does not Serial.Print out the stuff on the void setup, only the void loop. Using the arduino IDE serial monitor to test this out. 

Hope to hear from you soon! Thanks!

CongducPham, 2016-11-15:
Well, your issue is strange. The CS pin is definitely 2 on the UNO. I 
don't see why it is not working.

The issue you have with the Pro Mini is also strange, maybe try to add 
some Serial.flush() statements in between.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-11-15:
Hello, 

Is it because the library that you used only allowed SX1272 & 76? As the transceiver I am using is the 78. 

Thanks!

CongducPham, 2016-11-15:
No, it also works with the SX1278 because some radio modules use 
indifferently SX1276 or SX1278. So I don't think it is the issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-11-15:
Uncommented 20DBM
// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
//
// uncomment if your radio is an HopeRF RFM92W or RFM95W
//#define RADIO_RFM92_95
// uncomment if your radio is a Modtronix inAirB (the one with +20dBm features), if inAir9, leave commented
//#define RADIO_INAIR9B
// uncomment if you only know that it has 20dBm feature
#define RADIO_20DBM

Changed x to M
  // Select output power (Max, High or Low)
#if defined RADIO_RFM92_95 || defined RADIO_INAIR9B
  e = sx1272.setPower('M');
#else
  e = sx1272.setPower('M');
#endif 

That's all I have to uncomment right? Other than that the ping pong code remains the same?

CongducPham, 2016-11-15:
This is to set the correct output power whether your radio module uses 
PA_BOOST or not. But first the transceiver must be detected, so this is 
not the issue here.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kenchia10, 2016-11-16:
Alright thank you, will give it a try again. Thanks!