CongducPham/LowCostLoRaGw #46: Transceiver not recognized on HopeRF module

JasonTiel, 2016-10-24:
Hi Pham and all,
I'm trying to use your super nice lora library on a Anarduino Mini wireless with a LoRa RFM95W 915MHz tranceiver. http://www.anarduino.com/miniwireless/ . I have spent some time to dig into your SX1272 h and cpp files. So far I get a 'Unrecognized transceiver' message when I'm running the pingpong end node example. The output from the version variable, version = readRegister(REG_VERSION); I'm getting is 0x00, instead of the 0x22 or 0x12 . The boards I have looks similar to the Nexus configuration, with the hopeRF module on the back of a small arduino footprint.  Do you think there is hope for me to get this working considering the hardware I have? I have a few nodes up and running using the RadioHead library, but I much rather implement what you have built.

Connections between the LoRa transceiver and ATmega328p on the Mini wireless anarduino version.
RFM95W -> 328p
MISO -> D12
MOSI -> D11
SCK -> D13
NSS -> D10
RESET -> n/c
DIO5 -> D4 Ready
DIO2 -> n/c
DIO1 -> n/c
DIO0 -> D2 (INT0 interrupt pin) Rx done
DIO4 -> n/c
DIO3 -> n/c

I have separated out the hopeRF module from the mini wireless PCB, and set it up on a breadboard. I have verified the functionality of this breadboard setup with the RadioHead library. Now I have only connected the SPI pins, vcc, and ground. 

The transceiver is never recognized, but I forced the driver to set the SX1276 settings to see what would happen (I have also tried to force it to SX1272).

The pingpong serial output looks like:

Simple LoRa ping-pong with the gateway
Arduino MINI/Nexus detected

Starting 'ON'
## Reading:  ##	Register 42:  0
## Reading:  ##	Register 42:  0
Unrecognized transceiver, Version found is: 0
Trying SX1276 settings...: SX1276 LF/HF calibration
## Writing:  ##	Register 9:  0
## Reading:  ##	Register 3B:  0
## Writing:  ##	Register 3B:  40
## Reading:  ##	Register 3B:  0

Starting 'setChannel'
## Reading:  ##	Register 1:  0
## Writing:  ##	Register 1:  1
## Writing:  ##	Register 6:  D9
## Writing:  ##	Register 7:  0
## Writing:  ##	Register 8:  0
## Reading:  ##	Register 6:  0
## Reading:  ##	Register 7:  0
## Reading:  ##	Register 8:  0

Starting 'isChannel'
## Writing:  ##	Register 1:  0
## Reading:  ##	Register 3B:  0
## Writing:  ##	Register 3B:  40
## Reading:  ##	Register 3B:  FF
## Reading:  ##	Register 3B:  0

Starting 'setMaxCurrent'
## Reading:  ##	Register 1:  0
## Writing:  ##	Register 1:  1
## Writing:  ##	Register B:  3B
## Writing:  ##	Register 1:  0
## Setting ON with maximum current supply ##

Starting 'setLORA'
## Writing:  ##	Register 1:  0
## Writing:  ##	Register 1:  80
## Writing:  ##	Register 1:  81
## Reading:  ##	Register 1:  0
...

I find the RESET pin confusing, since it's defined in the driver code, but sometimes I see in the instructions and advice that the RESET pin is not necessary? Originally pin 3, I moved it to arduino pin 8, as to not occupy a pin with interrupt capabilities. 

I'm using the ARDUINO_AVR_MINI setting to get the SS pin on D10. ie. #define SX1272_SS 10

Am I missing something here? Is this transceiver really compatible with your driver code?  The text on the microchip on the HopeRF module is: RF96 1411 W2R235, and works fine with the Radiohead library and RHReliableDatagram.h,  RH_RF95.h .

Grateful for advice!

CongducPham, 2016-10-24:
Well, for sure the lib works for the RFM95W as we do have tested them. With our lib, you do not need to connect any DIO, although I don't think that it is the issue here. However, try to leave all DIO pin unconnected. So if you connected as you said:

MISO -> D12
MOSI -> D11
SCK -> D13
NSS -> D10
I really don't see why it is not working.

Duc.

JasonTiel, 2016-10-24:
Oh!! I found it. 
It was not changing the SS pin to 10.
Seems like the #define statement in the arduino program  (#define ARDUINO_AVR_MINI) did not carry over to the SX1272.h row  53, #ifdef 

#if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_NANO || defined ARDUINO_AVR_MINI || defined __MK20DX256__
#define SX1272_SS 10
#else
#define SX1272_SS 2
#endif

So  the pin was always setup on 2. 
It seems to work with the other DIO ports connected as well, so I don't have to modify the boards. as far as I can tell now.

CongducPham, 2016-10-25:
Well, for sure the lib works for the RFM95W as we do have tested them. 
With our lib, you do not need to connect any DIO, although I don't think 
that it is the issue here. However, try to leave all DIO pin 
unconnected. So if you connected as you said:

MISO -> D12
MOSI -> D11
SCK -> D13
NSS -> D10

I really don't see why it is not working.

Duc.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|