CongducPham/LowCostLoRaGw #47: Compatibility with Lora semtech SX1272

lucasf100, 2016-11-04:
Hi,
Im using a Semtech LORA SX1272,
I already load the program on my arduino Uno but its not reconized by the programm,
How can we fix it ?

Friendly
Lucas.E

CongducPham, 2016-11-04:
Normally it is supported what are the error messages? Check the SPI 
pins. What is your board?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

lucasf100, 2016-11-04:
Hi, 
The errors is : " the card is not supported "
My board is an arduino UNO
I ll check the pins also
Regards

CongducPham, 2016-11-04:
I don't know what sketch you are running. I'm not sure that this message 
comes from the communication library nor the sketch examples.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

lucasf100, 2016-11-05:
Hi,
I got this : 
"LoRa temperature sensor, extended version
Unrecognized transceiver
...
... "

CongducPham, 2016-11-05:
On the Uno, the CS pin should be connected to digital pin 2. Please 
check the SPI connections.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

lucasf100, 2016-11-05:
Yes ! I forgot that,
I try and I tell you if its working
Regards
Lucas

lucasf100, 2016-11-07:
Hi,
Now I got this :
" LoRa temperature sensor, extended version
SX1272 detected, starting
...
...
...
...
...
..."
I need to wait the loading or I miss something ?

Regards 
Lucas

CongducPham, 2016-11-07:
What is exactly your lora radio module? It is an Libelium SX1272? Seems that the connection is now ok, but the radio is unstable.

lucasf100, 2016-11-07:
The Radio Module is a Semtech SX1272,
[http://images.google.fr/imgres?imgurl=https%3A%2F%2Fdeveloper.mbed.org%2Fmedia%2Fcomponents%2Fpinouts%2FxPictureOfficial.jpg.pagespeed.ic.FzyoBOGJLz.jpg&imgrefurl=https%3A%2F%2Fdeveloper.mbed.org%2Fcomponents%2FSX1272MB2xAS%2F&h=1104&w=1150&tbnid=3gCVY6zRPZvbKM%3A&vet=1&docid=kVSZ-4WGi5UHvM&ei=CaogWMq1HoKVUau_juAL&tbm=isch&iact=rc&uact=3&dur=897&page=0&start=0&ndsp=9&ved=0ahUKEwiK3r3UhpfQAhWCShQKHaufA7wQMwgeKAEwAQ&bih=490&biw=1024](url)

CongducPham, 2016-11-07:
So you just have to connect the SPI pins (MOSI, MISO, NSS and SCK) and the 3.3 and GND pins, nothing else. NSS must be on digital 2. For MISO, MOSI and SCK, either use the ICSP header or SCK to pin 13, MOSI to pin 11 and MISO to pin 12 on your Uno.

lucasf100, 2016-11-08:
Hi,
I have connected the pins, and I ve got this message:
"LoRa temperature sensor, extended version
Unrecognized transceiver"
Now the card is unknow :/

lucasf100, 2016-11-08:
Im trying something else

lucasf100, 2016-11-08:
Now Its working but Only on the ICSP! not on the Digital pin of the arduino
Regards
Lucas

lucasf100, 2016-11-08:
It Worked only once time,
after a succes try I connect a temperature sensor on the lora shield and after the card is didnt recognize like before....
Now even if I try with out captor its still not working
I got this : 
"LoRa temperature sensor, extended version
Unrecognized transceiver"
Regards

CongducPham, 2016-11-08:
So maybe, try to comment LOW_POWER to not use low power. Maybe that is 
an issue with the LoRa card that you are using.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

lucasf100, 2016-11-10:
Hi,
I try to uncomment but still not working :(

lucasf100, 2016-11-10:
Its working :)

CongducPham, 2016-11-11:
Great! Good to hear!

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ccalas, 2016-12-12:
Hi Lucas ! I'm having the same problem.. Could you please tell me how you finally connect the semtech module to the Uno ?

Colin

lucasf100, 2016-12-12:
Hi,
You need to connect to the ICSP on the arduino

Lucas

ccalas, 2016-12-12:
Hi, thanks for your reply ! Anyway it worked without de ICSP header.

Colin