CongducPham/LowCostLoRaGw #48: RPI dont receive message

paula75, 2016-11-04:
Hello!

First of all thank you for your code and tutorials, there are really helpful.
I am using a RPI 3 for the gateway and an arduino uno for the end device. The module is inAir9. I have follow the instruction for sending a message from the node to the gateway, but the gateway is not receiving the message.
I changed the mode in Arduino_LoRa_Simple to mode 4. Also i am using a 915MHz antenna, but i am using 868 MHz in the transmission, as i understand this should work to, but to a very short range, i have the modules to 20 cm.

This is what i get from the node:
"Simple LoRa temperature sensor
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 22
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 533
Temp is 260.25
Sending \!#4#TC/260.25
Real payload size is 14
--> CAD duration 138
OK1
--> waiting for 1 CAD = 16
--> CAD duration 138
OK2
LoRa pkt size 18
LoRa pkt seq 22
LoRa Sent in 580
LoRa Sent w/CAD in 718
Packet sent, state 0
Switch to power saving mode
Successfully switch LoRa module in sleep mode
............................................."

And from the gateway i get this:
"Unrecognized transceiver.
...
...
***Power ON: state 0
Default sync word: 0x8
LoRa mode 4
Setting mode: state 1
Channel CH_10_868: state 0
Set LoRa Power to x
Power: state 1
Get Preamble Length: 255
LoRa addr 1: state 0
SX1272/76 configured as LR-BS. Waiting RF input for transparten RF-serial bridge"

I hope you  could help me to solve this problem,
Regards

CongducPham, 2016-11-05:
Hello, first of all, if you using inAir9, then check that all lines in 
the arduino sketch are commented:

//#define RADIO_RFM92_95
// uncomment if your radio is a Modtronix inAirB (the one with +20dBm 
features), if inAir9, leave commented
//#define RADIO_INAIR9B
// uncomment if you only know that it has 20dBm feature
//#define RADIO_20DBM

then on the RPI, for radio.makefile

#CFLAGS=-DRADIO_RFM92_95
#CFLAGS=-DRADIO_INAIR9B
#CFLAGS=-DRADIO_20DBM

Then not sure that your setting is optimal: 915Mhz antenna. Why don't 
you just use 915MHz band? Also, why mode 4? If you switch to mode 4, 
then both devive and gateway must be on this mode.

I see from the RPI output that there are some issue when starting the 
radio module. How dod you install the whole package? from scratch or 
using the provided SD card image? How do you launch the gateway? 
manually or on startup of the RPI?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

paula75, 2016-11-07:
Thanks for your answer
The lines in the arduino are commented, in the RPI the first line wasn't commented(#CFLAGS=-DRADIO_RFM92_95), i fix it, but still not work.
I was setting mode 4 because it was for default in the RPI, so i changed the mode in the arduino. I wanted to test the code in the 868 band for make it work and the i would change it to 915MHz band. But i supposed that i can change it now. 
In arduino i commented 
#define BAND868 and uncomment the #define BAND900 line and change to mode 5(for what i looked in the RPI file lora_gateway.cpp the modes for band 900MHz are 0-12, and for the 868MHz are 10-18 is that correct?).

I install the gateway with the SD card image. After i changed the band and the mode 5. In the terminal i recompiled using make lora_gateway and the run it with sudo ./lora_gateway

CongducPham, 2016-11-07:
> Thanks for your answer
> The lines in the arduino are commented, in the RPI the first line
> wasn't commented(#CFLAGS=-DRADIO_RFM92_95), i fix it, but still not work.
> I was setting mode 4 because it [...]
LoRa mode are not the same than frequency channel. In 900MHz band the 
default channel is indeed CHANNEL_05_900 which 913.88. LoRa mode defines 
a combination of BW and SF. See 
https://github.com/CongducPham/LowCostLoRaGw#annexa

So you have to be on the same channel and use the same mode. In your 
case, don"t change the mode to 5 then.
>
> I install the gateway with the SD card image. After i changed the band 
> and the mode 5. In the terminal i recompiled using make lora_gateway 
> and the run it with sudo ./lora_gateway

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

paula75, 2016-11-07:
Thanks a lot for your help. 

After i tried different things finally worked, the gateway received a message. But when i tried to send several message, i changed the variable "unsigned int idlePeriodInMin = 5;" for send more often, but sometimes the arduino can't switch to sleep mode and do not send the message. Is there a restriction for the time between messages?

CongducPham, 2016-11-08:
I don't think that there is restriction embedded in the radio as it is 
not a LoRaWAN stack. So the issue may be somewhere else.

You have to know however that sub-GhZ transmissions are usually 
regulated in power and duty-cycle, depending on your countries.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

paula75, 2016-11-18:
Hello

A question: Is it normal that the connection of the gateway and the end device late many times for to be correct? 
Sometimes i have to make several times "sudo ./lora_gateway" for the RPI configures properly, is it normal? or it might be a problem with interference o something?

Regards,

CongducPham, 2016-11-18:
No, it is not normal. Sometime you may miss a packet, but not many. If 
you have to run several time the program, maybe the radio module is 
unstable, check connection pin. What radio module are you using?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

paula75, 2016-11-18:
I am using inAir9, i tested the same module on an arduino for gateway and it worked perfectly. The problem is with the RPI, i checked the connections pines and it should be ok, The CS pin of the module should be on the SPI_CE0 of the RPi right?

For the gateway i download the SD card image and run it. But i'm going to try run it manually

CongducPham, 2016-11-21:
Could you provide the output from the RPI when you say that you have to "make several times "sudo ./lora_gateway" for the RPI configures properly"?