CongducPham/LowCostLoRaGw #49: Issue with Pingpong test (others?) when gateway disconnects?

bcsteeve, 2016-11-11:
I'm brand new to this, so I'm not sure how helpful this report will be.

I got everything working and ran the Pingpong test.  It worked!  Yay.  But as a quick test I disconnected the gateway.  As expected, it reported "no pong".  But then when I re-connected the gateway... it no longer says "no pong" but it also doesn't report a pong.  It just goes into a continual "ping sent" with no feedback at all.  Resetting the end device fixes it.  So it seems like there's some state that it is put in once "no pong" is returned that no puts it into a practical (not true) hang state.

Probably just a a software bug in the pingpong test.  If true, that's no big deal as that is really just a proof of functionality test.  But could it point to a larger robustness problem where the end devices have to be reset any time the gateway goes down?  I'd hope not.

CongducPham, 2016-11-11:
Hi, what do you mean by "you disconnected the gateway" and 
"reconnected"? Did you power off the gateway? You unplugged the radio 
pins on the gateway?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

bcsteeve, 2016-11-11:
I tried two things on the gateway:  1) unplug the USB (power); 2) press the reset button.  Both result in the same thing... it loops in a "sent Ping" until I reset the end device.

CongducPham, 2016-11-12:
Thanks for the information. I will try to reproduce the error and see 
what is the issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2016-11-16:
Hi, there was actually a bug in the SX1272 lib. So hust get the latest 
version of SX1272.cpp and SX1272.h in the Arduino folder. Now the 
PingPong example should work fine, even when deconnecting, then 
reconnecting gateway.

Thanks for pointing out  the issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

bcsteeve, 2016-11-17:
No problem.  So now I downloaded the whole project again but this time it won't compile.  I get this error:

`Arduino: 1.6.12 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\Steven\Downloads\LowCostLoRaGw-master\Arduino\Arduino_LoRa_Gateway\Arduino_LoRa_Gateway.ino: In function 'void startConfig()':

Arduino_LoRa_Gateway:556: error: 'class SX1272' has no member named 'setPowerDBM'

   e = sx1272.setPowerDBM((uint8_t)MAX_DBM);

              ^

exit status 1
'class SX1272' has no member named 'setPowerDBM'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

CongducPham, 2016-11-17:
SX1272.cpp and .h has been updated. Please get the latest version.

bcsteeve, 2017-01-27:
This bug either crept back in, or was never resolved.  I'm now on the latest and I re-discovered the same issue.  Power on end device and power on gateway... OK.  Now disconnect/reconnect gateway (either by pulling power or pressing reset button) but leave end device alone... the devices will no longer communicate until the end device is also reset.

This is with the PingPong code on the end device and Arduino Gateway on the gateway.

bcsteeve, 2017-01-27:
[removed:  I was wrong.  After several minutes communication resumed]