CongducPham/LowCostLoRaGw #50: rasp can't rcv data ,but show the valid header ???

lancezhangsf, 2016-11-21:
Hi,CongducPham

I open the debug,i fond that the rasp rcv the data ,it show the valid Header ,but can't rcv the data :+1: 
......................................................
## Reading:  ##	Register 12:  0
## Reading:  ##	Register 12:  0
## Reading:  ##	Register 12:  0
## Reading:  ##	Register 12:  50
## Valid Header received in LoRa mode ##
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0

....................................................

can you give me some advie???  thank you very much.

CongducPham, 2016-11-21:
You need to give me more details on your setup.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

lancezhangsf, 2016-12-09:
root@raspberrypi:/home/pi/WorkSpace/Raspberry LORA# ./lora_gateway 

Starting 'ON'
## Reading:  ##	Register 42:  0
Version:0
## Reading:  ##	Register 42:  12
SX1276/7/8/9 detected, starting.
## Writing:  ##	Register 9:  0
## Reading:  ##	Register 3B:  82
## Writing:  ##	Register 3B:  C2
## Reading:  ##	Register 3B:  0
## Reading:  ##	Register 3B:  0
## Writing:  ##	Register 3B:  40
## Reading:  ##	Register 3B:  0

Starting 'setMaxCurrent'
## Reading:  ##	Register 1:  0
## Writing:  ##	Register 1:  81
## Writing:  ##	Register B:  3B
## Writing:  ##	Register 1:  0
## Setting ON with maximum current supply ##

Starting 'setLORA'
## Writing:  ##	Register 1:  0
## Writing:  ##	Register 1:  80
## Writing:  ##	Register 1:  81
## Reading:  ##	Register 1:  81
...
## LoRa set with success ##

Starting 'getPreambleLength'
## Reading:  ##	Register 20:  0
## Reading:  ##	Register 21:  8
## Preamble length configured is 8 ##

Starting 'setSyncWord'
## Reading:  ##	Register 1:  81
## Writing:  ##	Register 1:  81
## Writing:  ##	Register 39:  12
## Reading:  ##	Register 39:  12
## Sync Word 12
 has been successfully set ##
## Writing:  ##	Register 1:  81

Starting 'getSyncWord'
## Reading:  ##	Register 39:  12
## Sync word is 12 ##
^$**********Power ON: state 0

Starting 'getSyncWord'
## Reading:  ##	Register 39:  12
## Sync word is 12 ##
^$Default sync word: 0x12
^$LoRa mode 1

Starting 'setMode'
## Reading:  ##	Register 1:  81
## Writing:  ##	Register 1:  81

Starting 'setCR'
## Reading:  ##	Register 1:  81
## Writing:  ##	Register 1:  81
## Reading:  ##	Register 1D:  72
## Writing:  ##	Register 1D:  72
## Reading:  ##	Register 1D:  72

Starting 'isCR'
## Coding Rate 1 has been successfully set ##

## Writing:  ##	Register 1:  81
[... 175 lines cut ...]
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
## Reading:  ##	Register 25:  0
。。。。。。。。。。。。。。。。。。
。。。。。。。。。。。。。

CongducPham, 2016-12-13:
Could you give me the details of the setup: what boards are you using, what is the radio module, what example are you running,...