CongducPham/LowCostLoRaGw #52: Ping-Pong - Waiting for DIFS problem

ciclonite, 2016-12-04:
Hi CongDuc,
I've a problem with ping pong sketch.
Configuration ;
Remote
- Arduino NANO V3
- LoraBee V1.1 chip RFM95w
GW
Raspberry Pi2
LoraMode 1

When i start the serial i see  this

###1
--> waiting for 5 DIFS (DIFS=3SIFS) = 2745
--> retry
--> CAD duration 60
###1
--> waiting for 4 DIFS (DIFS=3SIFS) = 2196
--> retry
--> CAD duration 61
###1
--> waiting for 7 DIFS (DIFS=3SIFS) = 3843

Or similar ...

On GW 
Destination: 8
Source: 1
ACK number: 6
ACK length: 2
ACK payload: 0
ACK SNR last rcv pkt: 5
 ##

--> CAD duration 201
OK1
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> RSSI 127
--> CAD duration 201
OK1
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> RSSI 126
--> CAD duration 201
OK1

Thank a lot.

P.S. With the sensor example Sketch work propely.

CongducPham, 2016-12-05:
It may be a power problem.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ciclonite, 2016-12-05:
Thanks for reply,
I've connected the arduino nano directly to PC usb, today i will check with a multimeter the output from 3v3 and the 5v.
Thanks a lot,
Giovanni

selvamaniraj, 2016-12-21:
Hi ConducPham
 I have a problem with ping pong sketch. I am using arduino mini with Doorji lora sx1278. 
SPI Wiring Connections 
Arduino mini       Doorji LoRa
    10                        D5
    11                       MOSI
    12                       MISO
    13                        SCK
    and also i give 3.3v power supply for this board . finally i am upload the program. and it shows 

Simple LoRa ping-pong with the gateway
Arduino Pro Mini detected
Unrecognized transceiver
...
...
...
...
...
...
 Can you help me with this issue and thanks in advance.