CongducPham/LowCostLoRaGw #53: Could not switch LoRa module in sleep mode

ciclonite, 2016-12-04:
Hi CongDuc,
I reiceve this error when use the example sketch generic sensor.
Setup:
RFM95W (lora-bee v 1.1)
arduino NANO v3
DHT22
The module was unable to go in sleep mode.
Thanks,
Giovanni.

CongducPham, 2016-12-05:
Did you have this issue every time?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ciclonite, 2016-12-05:
Hi Congduc,
Thanks for reply.
Yes, every time.
Thanks,
Giovanni

escyes, 2017-03-25:
I just have the same isue when I try InAir9, not allways, it's seems to go better wen I connect the RESET pin of the module

CongducPham, 2017-03-25:
Well, it happen sometimes and I haven't figured out why, it is a bit 
random. I don't thing that the RESET is going to change something actually.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

escyes, 2017-03-25:
Now it seems go well when I put that delay(100) in sx1272.cpp

_int8_t SX1272::setSleepMode() {
    int8_t state = 2;
    byte value;

    writeRegister(REG_OP_MODE, LORA_STANDBY_MODE);
	delay(100);  //inserted to avoid REG_OP_MODE stay = 0x40 (no sleep mode)
    writeRegister(REG_OP_MODE, LORA_SLEEP_MODE);    // LoRa sleep mode_

CongducPham, 2017-03-26:
Great, thanks, I will try it and integrate it.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

escyes, 2017-03-26:
Is just a small contribution to your great work, for which I am very grateful.
Best regards