CongducPham/LowCostLoRaGw #57: Arduino mini with Doorji sx1278 lora shows a error

selvamaniraj, 2016-12-21:
Hi ConducPham
I have a problem with ping pong sketch. I am using arduino mini with Doorji lora sx1278.
SPI Wiring Connections

Arduino mini - Doorji LoRa
10 - D5
11 - MOSI
12 - MISO
13 - SCK

and also i give 3.3v power supply for this board . finally i am upload the program. Finally it shows
a error in Serial Monitor Window

"Simple LoRa ping-pong with the gateway
Arduino Pro Mini detected
Unrecognized transceiver
...
...
...
...
...
..".
Can you help me with this issue and thanks in advance.

CongducPham, 2016-12-21:
Hi, why are you connecting D5? You should connect the CS pin of your Doorji.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

selvamaniraj, 2016-12-21:
Thank you for replaying me sir. I change the pin number  D5  to sck sir but it also shows same error sir.

CongducPham, 2016-12-21:
The CS pin of your module should be connected to pin 10 of your mini.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

selvamaniraj, 2016-12-21:
Sorry for the inconvenience .I am already connected  correct pin but it also working same sir.

CongducPham, 2016-12-21:
Give me a picture of your radio module because some Doorji modules are 
UART-based and not SPI-based.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

selvamaniraj, 2016-12-21:
I am using Doorji sx1278f LoRa Module, (not Doorji lora modem) and also i am attached the Picture of the LoRa module with datasheet.
[image]

[DRF1278F.pdf](https://github.com/CongducPham/LowCostLoRaGw/files/666830/DRF1278F.pdf)

CongducPham, 2016-12-21:
OK, you have the following setting:

10 - NSS
11 - MOSI 
12 - MISO 
13 - SCK
3.3v

Sorry to ask, did you connected the GND? Because I don't see the issue here since all pin connections seems correct.

selvamaniraj, 2016-12-22:
Yes I have connected my GND and my connections are as follows
Pin connections:
Arduino pro mini -Doorji LoRa 
10-NSS
11-MOSI
12-MISO
13-SCK
3.3v-VCC(8)
GND-GND(9)

CongducPham, 2016-12-22:
Not sure to understand what you meant by 

3.3v-VCC(8)
GND-GND(9)

Why pin 8 and pin 9?  There are the VCC and GND pins on the board where you should connect the corresponding radio module pins. Also, what version of the Pro Mini do you have? Because you need the 3.3v version, otherwise your board is in 5v which is not good for the radio module.

selvamaniraj, 2016-12-23:
Sorry for late reply.The board is working good. Thanks for your support.

Also i have another issue. The issue is that I have adafruit feather mo board (atsamd21g18a). The program shows a error as SerialUSB error.  So if change the version of arduino ide from 1.7.5 and 1.7.8 and tried uploading the program but I cant upload it. Can you help me??

CongducPham, 2016-12-23:
OK, I see the problem, I will correct it. To fix it quickly, find in the 
sketch the following statement:

#if defined __SAMD21G18A__

and

#ifdef __SAMD21G18A__

there should be at 2 places replace both with

#if defined __SAMD21G18A__ && not defined ARDUINO_SAMD_FEATHER_M0

I actually don't have the Feather M0, the Arduino Zero or M0 uses 
SerialUSB while the Feather M0 seems to use Serial.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2016-12-23:
Actually there are several places where I used

#if defined __SAMD21G18A__

So, the 2 places where you have to replace are:

fisrt:

// we wrapped Serial.println to support the Arduino Zero or M0
#if defined __SAMD21G18A__
#define PRINTLN                   SerialUSB.println("")

second:

#ifdef __SAMD21G18A__
   SerialUSB.begin(38400);
#else
   Serial.begin(38400);
#endif

For these 2 defines statements, replace by:

#if defined __SAMD21G18A__ && not defined ARDUINO_SAMD_FEATHER_M0

--------

OK, I see the problem, I will correct it. To fix it quickly, find in the 
sketch the following statement:

#if defined __SAMD21G18A__

and

#ifdef __SAMD21G18A__

there should be at 2 places replace both with

#if defined __SAMD21G18A__ && not defined ARDUINO_SAMD_FEATHER_M0

I actually don't have the Feather M0, the Arduino Zero or M0 uses 
SerialUSB while the Feather M0 seems to use Serial.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|http://liuppa.univ-pau.fr/        |http://www.univ-pau.fr    |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr        http://www.univ-pau.fr/~cpham     |
|----------------------------------------------------------------|