CongducPham/LowCostLoRaGw #59: Never switch on sleep mode

Sasgho, 2016-12-25:
happy christmas !!!

I connect one mini arduino with rfm95w, but never switch on sleep mode

any solution ?

regards

Simple LoRa temperature sensor
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 876.56
29.8
(Temp is 42.9
˜Sending \!#4#TC/29.7
Real payload size is 12
--> CAD duration 301
###1
--> waiting for 1 DIFS (DIFS=3SIFS) = 549
--> retry
--> CAD duration 301
###1
--> waiting for 1 DIFS (DIFS=3SIFS) = 549
--> retry
--> CAD duration 301

CongducPham, 2016-12-25:
You may have a current issue. How do you power the radio module? It is a 
Pro Mini at 3.3v version?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2016-12-28:
hi !

yes, my arduino mini works with 3.3 v

regards

Sasgho, 2016-12-28:
hi CongducPham !

i fixed this problem...

Simple LoRa temperature sensor
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 6
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 1517.87
29.9
(Temp is 52.5
52.5Sending \!#4#TC/52.5
Real payload size is 12
--> CAD duration 548
OK1
--> waiting for 1 CAD = 62
--> CAD duration 548
OK2
LoRa pkt seq 6
LoRa Sent in 5315
LoRa Sent w/CAD in 5863
Packet sent, state 1
Switch to power saving mode
Successfully switch LoRa module in sleep mode
...
----------------------------------------------------------
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
but this parameters are 0...it's ok ?

regards

CongducPham, 2016-12-28:
When state is 0 it means OK. I see that the transmission of the packet 
gives state=1.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2016-12-28:
hi CongducPham !

what's mean of the packet
gives state=1. ?

any problem ?

regards

gives state=1. ?

any problem ?

regards

CongducPham, 2016-12-30:
Yes, state=1 means that there has been some issue, typically a tiemout 
because the transmission has not been completed correctly.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|