CongducPham/LowCostLoRaGw #61: How to push RSSI to Thingspeak channel

tiki30, 2017-01-07:
Hi,
is it possible to push the RSSI value of received data to a thingspeak channel ?, sample code would be appreciated.

Thanks.

CongducPham, 2017-01-11:
Hi, in your CloudThingSpeak.py script, you can use 
thingspeak_thingspeak_uploadMultipleData(data_array) instead of the 
current usage of thingspeak_uploadSingleData(data, second_data).

Then, in main, just add an RSSI field in ldata prior to process the 
string line. For instance, if your sensor is sending:

##TC/34/HU/78/O2/45 and you want to have the RSSI, you can just append 
it to the line:

ldata = ldata + '/RSSI/'+str(RSSI)

so that ldata would be

##TC/34/HU/78/O2/45/RSSI/-115

for instance. put that just after the line "RSSI=arr[6]"

hope that helps.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

tiki30, 2017-02-03:
Hi,
thank you for your reply, apologies for my late response. This worked, many thanks to you and your team for a great, well documented project.

closed 2017-02-03 by tiki30