CongducPham/LowCostLoRaGw #63: the gateway received broken information

Sasgho, 2017-01-10:
when send string concatenated e.g temperature and humidity... the gateway received broken information
[image]
 

the transmitter info

Get back previous sx1272 config
Using packet sequence number of 36
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 29.3
(Temp is 29.3
61.2Sending \# **!#4#H61.2#T29.3**
Real payload size is 16
--> CAD duration 549
OK1
--> waiting for 1 CAD = 62
--> CAD duration 549
OK2
1
\!#4#H61.2#T29.320
LoRa pkt seq 36
LoRa Sent in 1735
LoRa Sent w/CAD in 2284
Packet sent, state 0
Switch to power saving mode
Successfully switch LoRa module in sleep mode

my trasmitter in arduino mini pro and the gateway is raspberry pi 2 B, conncet the sensor SparkFunHTU21D
temperature and humidity....

any help please !!!

regards

CongducPham, 2017-01-11:
> raspberrypi_lowcostloragw

You have to use the following syntax:

*\!#4 <https://github.com/CongducPham/LowCostLoRaGw/issues/4>#H/61.2/T/29.3*

*regards,
*

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2017-01-11:
Thanks, friend !!!
regards!

Sasgho, 2017-01-11:
the problem persis friend !!!ards

reg

Sasgho, 2017-01-11:
the problem persis friend !!!

regards!

Sasgho, 2017-01-13:
can you help me please !!))

regards

CongducPham, 2017-01-13:
What is exactely the problem?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Sasgho, 2017-01-14:
was fixed friend !!!!
my question is where be  into raspberry pi  have the string about string received (node,temperature , etc)

maybe sprintf_buf  on lora_gateway_cpp ???
 
because want store in external data base e.g. usb dongle flash memory

regards !!!

closed 2017-01-16 by Sasgho