CongducPham/LowCostLoRaGw #64: help! I cann't download the raspberrypi-jessie-WAZIUP-demo.dmg

xiaohuasheng, 2017-01-11:
who can share the image raspberrypi-jessie-WAZIUP-demo.dmg to me?  I can download it from provided link, thank you very much!
my email is 775079852@qq.com

CongducPham, 2017-01-11:
> my email is 775079852@qq.com <mailto:775079852@qq.com>

Hi, I've tested the link and it is OK.

http://cpham.perso.univ-pau.fr/LORA/WAZIUP/raspberrypi-jessie-WAZIUP-demo.dmg.zip

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

xiaohuasheng, 2017-01-13:
Thank you! I've solved it with a download software .

closed 2017-01-13 by xiaohuasheng