CongducPham/LowCostLoRaGw #66: support SX1278?

xiaohuasheng, 2017-01-14:
I connect Raspberry 3 with the SX1278 radio module. 
My SX1278 radio module supports 433MHz, so I edit the global_conf.json:
[image]
Run the gateway, it displays:
[image]
but when I send data from a lora end-device, it can receive anything, it supports SX1278 or not?
moreover, as the sheet shows, mode 10's BW is 500, but if I set BW in global_conf.json, such as 433, which will work?
[image]
thanks a lot!

CongducPham, 2017-01-15:
> image

BW is not the frequency. So it is not the parameter that you have to 
change. Use the freq field to indicate a frequency. Please, have close 
look at the README:

https://github.com/CongducPham/LowCostLoRaGw#running-in-433mhz-band

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

xiaohuasheng, 2017-01-16:
Oh! Thank you very much.

xiaohuasheng, 2017-01-17:
@CongducPham Thans for your reply, I have read 433MHz part of README, and did that as it suggests, the result is:
[image]
[image]
I run the same command but have two results, why? and I still can't receive data from end-device, how to debug it?
also, I don't know the meaning of the tips like"Power: state 0", what "state 0" represents? 
Hope for your answer!

CongducPham, 2017-01-17:
state = 0 means everything is OK. Here, setting the frequency seems to give an error of -1 which should not happen in the latest version of the SX1272 lib. I suspect that you are using an old version. Get the latest version of both SX1272.cpp and SX1272.h on the github and retry.

Anyway, normally, even if the returned state is -1, the gw should be running on 433.3MHz. If you don't receive anything, check if your end-device is on that frequency. What example are you using? Make sure that you also have BAND433 uncommented in the example.

chijiao, 2018-10-29:
I'm tested, support perfect. please change mode to 1. i have two sx1278 model. one as gateway, one as node. can send and received.