CongducPham/LowCostLoRaGw #68: Can I conect the device to an another Gateway ?

lucasf100, 2017-01-20:
Hi,

Is it possible to connect the device to a MULTITECH LORA Gateway ?
The gateway is aking me for a password and stuff that I dont know how to configure on the device.
How to configure it ?

Friendly
Lucas.E

CongducPham, 2017-01-20:
Dear Lucas,

Our framework has limited LoRaWAN support. We are still testing the 
reception of packets from an end-device using our lib on a LoRaWAN 
gateway such as the multitech. Will keep you informed. The other way is 
alread validated: an mDot can send message to our low-cost gateway.

best regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

lucasf100, 2017-01-20:
Thank you for your quick answer :)