CongducPham/LowCostLoRaGw #69: Cannot see anything on RPi 3 which has a InAir9 connected.

openedhardware, 2017-01-24:
I just followed the tutorial of the pdf file and compiled the `lora_gateway` on my RPi after cloning this repo.
Unfortunately `sudo ./lora_gateway` does not show anything on the terminal.
Is there any way of checking the InAir9 module?

Of course, I checked the wiring for several times.

Thanks.

CongducPham, 2017-01-28:
Did you make for RPI3?

 > make lora_gateway_pi2

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

openedhardware, 2017-01-28:
Hey.
Thanks for your reply in weekend!

Sadly I am using RPi 3.

CongducPham, 2017-01-28:
RPI2 and RPI3 work the same, so on RPI2 & RPI3 you have to use

 > make lora_gateway_pi2

an alias will will created so just launch:

 > sudo ./lora_gateway

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

openedhardware, 2017-01-29:
Thanks a lot!
I got it working on my RPi.
By the way, received data are broken most of times and I cannot read.
[image]

Any idea?