CongducPham/LowCostLoRaGw #70: Arduino compile error in Arduino_LoRa_Ping_Pong

bcsteeve, 2017-01-27:
I haven't changed anything of the code other than defining FCC_US_REGULATION, PABOOST and BAND900.  Load up that file into IDE and click verify and I get:

exit status 1
'MAX_DBM' was not declared in this scope

bcsteeve, 2017-01-27:
Oh, probably because you haven't defined MAX_DBM when FCC_US_REGULATION is selected.

bcsteeve, 2017-01-27:
grr.  with that fixed, now I have not enough memory errors.  Was this not designed to work on Arduino Pro Mini??  That's what the tutorial states, but the current code for this simple Ping Pong doesn't fit.

CongducPham, 2017-01-27:
Hi, you probably have check the version of your arduino lib, or try to 
recompile several time. Sometimes, there are bugs in the Arduino IDE.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

bcsteeve, 2017-01-27:
It probably also helps if I choose the proper Atmega 328 chip instead of 168 :)

My dumb, sorry... but the initial post in this issue stands.  It won't compile if FCC_US_REGULATION is selected.

CongducPham, 2017-01-27:
Yes, you were right, I will add a MAX_DBM for FCC.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

bcsteeve, 2017-01-27:
Stuck again... and confused.  I had this working before Christmas (when I put it down because I got too busy).  Today I first tested that ping pong worked (it did) then I uploaded latest of everything from GIT and now... I don't get "pong".  But I changed nothing (hardware) and the software is simply defining band, etc.

I get expected serial output from the end device:

> Simple LoRa ping-pong with the gateway
> Arduino Pro Mini detected
> SX1276 detected, starting
> SX1276 LF/HF calibration
> ...
> Setting Mode: state 0
> Setting Channel: state 0
> Setting Power: state 2
> Setting node addr: state 0
> SX1272 successfully configured
> --> CAD duration 549
> OK1
> --> waiting for 1 CAD = 62
> --> CAD duration 551
> OK2
> --> RSSI -124
> Sending Ping
> wait for ACK
> no ACK
> Packet sent, state 3
> No Pong!

the gateway side (which used to send ACK) doesn't give me any particular error:

> SX1276 detected, starting
> SX1276 LF/HF calibration
> ...
> ^$**********Power ON: state 0
> ^$Default sync word: 0x12
> ^$LoRa mode 1
> ^$Setting mode: state 0
> ^$Channel CH_5_900: state 0
> ^$Use PA_BOOST amplifier line
> ^$Set LoRa power dBm to 20
> ^$Power: state 2
> ^$Get Preamble Length: state 0
> ^$Preamble Length: 8
> ^$LoRa addr 1: state 0
> ^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge

To be honest, I don't remember what it used to read so I'm not sure if that "Waiting RF input" message is normal?  It never seems to get the "ping" and it never responds.

Any idea what I'm missing here?  Should I try to go back to your code from ~November?

Full disclosure:  I'm using an UNO for the Gateway.  I know you said that isn't sufficient, but that's what I have and it worked before, at least for this very basic Ping Pong test.