CongducPham/LowCostLoRaGw #71: no data running on 433MHz

mogyoros, 2017-02-07:
Dear Pham!

Thank you for all your efforts to share your experiences with LoRa technology.
Using the provided documentation and software I managed to create a working setup on 868MHz with a RPI3+[connection bridge](https://www.cooking-hacks.com/documentation/tutorials/raspberry-pi-to-arduino-shields-connection-bridge/)+[Libelium SX1272 module](https://www.cooking-hacks.com/sx1272-lora-module-for-arduino-raspberry-pi-intel-galileo-868-mhz) as gateway, and arduino UNO + [multiprotocol shield](https://www.cooking-hacks.com/multiprotocol-radio-shield-board-for-arduino-rpi-intel-galileo)+[Libelium SX1272 module](https://www.cooking-hacks.com/sx1272-lora-module-for-arduino-raspberry-pi-intel-galileo-868-mhz) as node. Using the LM35 temp sensor I was able to send data to thinkspeak.

Now I wanted to make a trial on 433MHz with different radio modules, and [followed ](https://github.com/CongducPham/LowCostLoRaGw#running-in-433mhz-band) this guide.
On the gateway side I have changed the antenna, and tried to compile the lora_gateway.cpp with #define BAND433 option, but that failed. Going through the library, I see the 433MHz options are missing. I am using the RPI jessie image that is published on the webpage. So I downloaded the git version and using the gw_full_latest to run the gateway.

> root@raspberrypi:/home/pi/LowCostLoRaGw/gw_full_latest# make lora_gateway_pi2
> g++  -DRASPBERRY -DRASPBERRY2 -DIS_RCV_GATEWAY -c lora_gateway.cpp -o lora_gateway_pi2.o
> g++ -c arduPi_pi2.cpp -o arduPi_pi2.o
> g++ -DRASPBERRY2 -c SX1272.cpp -o SX1272_pi2.o
> g++ -lrt -lpthread lora_gateway_pi2.o arduPi_pi2.o SX1272_pi2.o -o lora_gateway_pi2
> rm -f lora_gateway
> ln -s lora_gateway_pi2 ./lora_gateway
> root@raspberrypi:/home/pi/LowCostLoRaGw/gw_full_latest# ./lora_gateway
> lora_gateway      lora_gateway_pi2
> root@raspberrypi:/home/pi/LowCostLoRaGw/gw_full_latest# ./lora_gateway
> SX1272 detected, starting.
> ...
> ^$**********Power ON: state 0
> ^$Default sync word: 0x12
> ^$LoRa mode 1
> ^$Setting mode: state 0
> ^$Channel CH_0_433: state 0
> ^$Set LoRa power dBm to 14
> ^$Power: state 0
> ^$Get Preamble Length: state 0
> ^$Preamble Length: 8
> ^$LoRa addr 1: state 0
> ^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge

Seems everything ok.

On the node side, I have 2 arduinos, 1 nano (SS connected to PIN D10) and 1 uno (SS connected to PIN D2) chinese clones, and SX1276 (actually 1278) LoRa [modules](https://www.aliexpress.com/item/SX1276-SX3km-1278-433MHz-Wireless-Transceiver-Module-Long-Distance-Receiver-and-transmitter-LoRa-SPSP-Module/32711941138.html), also from china. Using the ping-pong template (#defined BAND433) the module starts fine, but no data is received on the gateway:

> Simple LoRa ping-pong with the gateway
> Arduino Nano detected
> SX1276 detected, starting
> SX1276 LF/HF calibration
> ...
> Setting Mode: state 0
> Setting Channel: state 0
> Setting Power: state 0
> Setting node addr: state 0
> SX1272 successfully configured
> --> CAD duration 550
> OK1
> --> waiting for 1 CAD = 62
> --> CAD duration 549
> OK2
> --> RSSI -127
> Sending Ping
> wait for ACK
> 

But if I change both gateway and node side to BAND866, the communication is established! (with a very bad SNR of course, due to the transmitter module is designed to operate on 433).
Also tried to run the gateway with --freq 433.3 option, but does not help.
Any idea what am I missing, since I am lost a bit now?
Thanks.
Kind regards,
Tamas

mogyoros, 2017-02-10:
Ok. I was to lame... SX1272 operates on High Band (860 - 1020 MHz) frequencies, while SX1278 is running on Low Band (137-175 & 410-525 MHz) frequencies only!
So yes, reading is always good before you buy something, SX1276/77/79 the one which is capable of running on both bands.
The strange thing still about the SX1278 is that, when configuring to operate on 868 MHz, it can produce a reasonable transmit power using PA_BOOST to be able to transmit data as a node. But when operating as a receiver on this band, it really fails...
Anyway, case is to be closed, solved myself

closed 2017-02-10 by mogyoros

CongducPham, 2017-02-12:
OK, great to hear that the issue was found. The design of the radio 
module, apart from the SX127X chip itself has great impact on the radio 
capabilitie as well, so issue like the one you describe may be explained 
by how the various amplifier lines are wired.

Wish you great success for your projects.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

liuxiaodong528, 2017-05-15:
Hello @CongducPham ,I use the sx1278 in the 433MHZ is ok,Now I would like to use 470MHZ how to modify the program?
Thanks ～～

CongducPham, 2017-05-15:
Just add a statement such as:

const uint32_t CH_470 = 0x758000 // 470MHz

and use this frequency for the end-device.

In the gateway, edit gateway_conf.json to have  "freq" : 470.0

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

liuxiaodong528, 2017-05-16:
Thanks for your help,i can use 470MHz to send and receive data,but I have some questions，Why is communication so close？How should I adjust the parameter to a greater distance？ 
Thanks @CongducPham

CongducPham, 2017-05-16:
Range depends on many factors (antenna, cables, placement,...), so hard 
to tell, regarding the lora parameters, so-called mode 1 with bw=125kHz 
and SF=12 normally provides the lowest sensibility thus the longest range.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|