CongducPham/LowCostLoRaGw #74: Gateway didn't received any packets

ramesh493, 2017-02-24:
Hello Pham,

I have installed OS on RPI3, I am not using your SD card image.

My gateway is connected to WIFI, i can log on to gateway with SSH.

I have installed all the files, and configured the gateway.

when i start my gateway using sudo ./lora_gateway_pi2 i am seeing the below output.

***Power ON: state 0
Default sync word: 0x12
LoRa mode 1
Setting mode: state 0
Channel CH_10_868: state 0
Set LoRa Power dBm 14
Power: state 0
Get Preamble Length: 8
LoRa addr 1: state 0
SX1272/76 configured as LR-BS. Waiting RF input for transparten
RF-serial bridge"

CongducPham, 2017-02-24:
So the gateway seems to work fine.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ramesh493, 2017-02-24:
I am not sure the gateway is working fine, why because,
i have started script ./cmd.sh and check for the 3 option, i only see one process running there

CongducPham, 2017-02-24:
For the inAir9B, check that PABOOST is defined.

For the gateway, try to run it mannually:

 > sudo python ./start_gw.py &

 > disown %1

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Nuno-Nico, 2017-02-27:
I am with the same issue. the device is recongnized but i fail to recive any data. I am using a RFM95. the arduino is sending correctly.

any suggestion?

CongducPham, 2017-02-27:
What is the sketch you are using at the device side?

CongducPham, 2017-02-27:
great to hear to it is fine now.

ramesh493, 2017-02-28:
its working now,

thanks

closed 2017-02-28 by ramesh493