CongducPham/LowCostLoRaGw #75: Issue with Adruino_Lora_Temp.ino Sketch

ramesh493, 2017-02-24:
Hi

LoRa temperature sensor, extended version
Arduino Nano detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 68
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 531
Temp is 259.28
Sending \!#3#TC/259.27
Real payload size is 14
\!#3#TC/259.27
plain payload hex
5C 21 23 33 23 54 43 2F 32 35 39 2E 32 37 
Encrypting
encrypted payload
4F 29 E7 49 77 A1 C2 E7 81 E1 01 BD 6A DB 
calculate MIC with NwkSKey
transmitted LoRaWAN-like packet:
MHDR[1] | DevAddr[4] | FCtrl[1] | FCnt[2] | FPort[1] | EncryptedPayload | MIC[4]
40 06 00 00 00 00 00 00 01 4F 29 E7 49 77 A1 C2 E7 81 E1 01 BD 6A DB 3F 7A AD CB 
end-device uses encapsulated LoRaWAN packet format only for encryption
--> CAD duration 300
###1
--> waiting for 1 DIFS (DIFS=3SIFS) = 549
--> retry
--> CAD duration 300
###1
--> waiting for 1 DIFS (DIFS=3SIFS) = 549
--> retry
--> CAD duration 301
###1
--> waiting for 3 DIFS (DIFS=3SIFS) = 1647
--> retry
--> CAD duration 301
###1
--> waiting for 7 DIFS (DIFS=3SIFS) = 3843

CongducPham, 2017-02-24:
The issue seems to be on the device side. What is the radoo module that 
you use?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ramesh493, 2017-02-24:
radio module is inAir9B

ramesh493, 2017-02-28:
Hi, 
i have checked all the connections again, its working fine now

Thanks

closed 2017-02-28 by ramesh493